ArduPilot/ardupilot #10190: Use "ground speed " for TECS Control

yuxiqun, 2019-01-07:
According to "ardupilot/libraries/AP_TECS/AP_TECS.cpp", it use “true airspeed” to calculate airplane's kenitic energy (SKE) and "height" to calculate airplane's potential energy(SPE). The sum of SKE and SPE is airplane's total energy.Why not use "ground speed" to calculate SKE?

magicrub, 2019-01-07:
because ground speed is not relevant. in high winds, your aircraft could in fact be flying with a negative ground speed and a positive (above stall) airspeed.

IamPete1, 2019-01-07:
I agree that when flying all that really matters is airspeed, but to calculate kinetic energy of the aircraft it makes scene to calculate using true ground speed. ie if you we flying at a the wind speed into the wind and had zero ground speed you would also have zero kinetic energy. 

Maybe its is taken relative to the 'air' reference frame rather than the ground. Although this is somewhat confusing, if your head wind were to suddenly stop you would still have to accelerate the aircraft.

Naterater, 2019-01-07:
The basis for discussion here is that wind changes the kinetic energy of the aircraft. In the earth-frame, that is correct. In the body-frame, it is not. We are concerned with the aircraft's body-frame trajectory in the air, not on the ground. TECS is essentially 1-dimensional (altitude/energy, not X-Y based). Therefore, ground speed is not relevant.

Let's provide an example to illustrate this. A plane is flying with a strong tailwind and is assumed to have additional kinetic energy beyond its normal flight envelope. When a climb is commanded, the plane calculates this ground-based kinetic energy and tells itself that no additional throttle is necessary to climb (it can exchange speed for altitude). The plane does not increase throttle as it pitches up, loses airspeed, and stalls. You can see that the additional earth-frame kinetic energy from groundspeed does no good.

The only validity for using ground-speed would be if we are turning into or out of the wind (state changes). Accelerations won't add up to the same change in airspeed. Since changing TECS to be 2 or 3-D calculation would be quite complicated, I think this issue should be closed.

IamPete1, 2019-01-07:
Thanks for the explanation, that is a good example. Quite hard to get your head around when we're used to standing on the ground watching aircraft, I guess from a pilots seat its more intuitive.

magicrub, 2019-01-07:
@IamPete1 When picturing the energies and frame bodies involved, I always find it helpful to replace "air" with "water". Moving river == wind. [This feature](https://github.com/ArduPilot/ardupilot/pull/2749) is where it really sunk in for me. (no pun intended)

closed 2019-01-07 by magicrub