ArduPilot/ardupilot #10194: Labels for Issues

Naterater, 2019-01-07:
### Feature request

**Is your feature request related to a problem? Please describe.**
The issue list is very long at over 1,000 issues, and there are many issues without tags

**Describe the solution you'd like**
I can go through most of them, and while I do, I am considering adding tags to intelligently sort them.

Most of the issues appear to fall into the following categories:
Feature Requests - New flight behavior
Feature Requests - Changes/enhancements to flight behavior
Feature Requests - New setup
Feature Requests - Changes/enhancements to setup
Issues - Specific Case
Issues - Simulation
Issues - Flight Behavior
Issues - Sensor/Auxillary hardware Behavior
Issues - Radios
Issues - Setup/Parameters
Issues - Identified in Code

Can we implement the following tags?
FeatureRequest
Issue
NewIdea
FlightBehavior
Setup
Specific
Hardware
Radio
CodeIdentified
BUGconfirmed
OLD (for cleanup purposes)

**Describe alternatives you've considered**
Applying Enhancement or BUG to almost every issue.

I'm open to suggestions. Maybe this belongs in the discuss.ardupilot.org.

rmackay9, 2019-01-08:
Thanks again for reviewing the issues.

I think adding the extra labels is useful is someone is actually going to use them to find issues but I'm less sure that people will actually do that.. if we don't use it then it will end up being extra effort without much payoff.

I do use the labels to find "Copter" "Bugs".. but I'm not sure I would use some of the new ones.  Maybe it would be useful to have labels for code subsystem.  So for example, "Compass" or "GPS".. although I think we often label that by prepending it to the front of the subject line.

WickedShell, 2019-04-05:
Going to close this as it seems to be just stale (and is contributing to the many issues). Overall I'm not a fan of the proposed tags as NewIdea overlaps entirely with FeatureRequest, Specific is terribly vauge, and OLD should either be closed or not, but fundamentally finding old issues is easy and not something you need a tag for. (You don't even need a tag to exclude them when searching).

If you want to discuss this further I'd suggest bringing it up as a DevCallTopic.

closed 2019-04-05 by WickedShell