ArduPilot/ardupilot #10196: Plane: Loiter Minimum Altitude Request

Naterater, 2019-01-08:
### Feature request

**Is your feature request related to a problem? Please describe.**
I had a user accidentally switch to loiter in the middle of an auto-takeoff. This caused a sweeping circle very low to the ground.

**Describe the solution you'd like**
New parameter WP_LOITER_ALTMIN
Display Name: Minimum Loiter Altitude
Description: Minimum altitude that the aircraft will loiter relative to home. Prevents low-altitude loiter events.

**Describe alternatives you've considered**
Remove the loiter option for my users.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Additional context**
[image]

magicrub, 2019-01-08:
What do you think the default value should be? Maybe 0? What do negative numbers mean?

Naterater, 2019-01-08:
Probably 0 (disable). Range 0-32,600 (or whatever that magic number is).

peterbarker, 2019-01-08:
On Mon, 7 Jan 2019, Nathan E wrote:

> Is your feature request related to a problem? Please describe.
> I had a user accidentally switch to loiter in the middle of an auto-takeoff. This caused a sweeping circle very low to the ground.

Seriously, your user could have selected a half-a-dozen other ways to kill 
their vehicle.  Why pick on Loiter?

Perhaps another option you might consider here is that if you are doing an 
automated takeoff that the Loiter will occur at your takeoff altitude.... 
fewer parameters....  not sure I really like the idea of Loiter doing 
anything other than making a circle - even if it means writing the 
incident off as "Crashing as Intended"....

ChrisBird, 2019-01-14:
@auturgy dont you normally have the geofence off for takeoff and landing?

auturgy, 2019-01-14:
It’s a good test case for fence_autoenable. If it was set to 1 it should activate when the launch completes, so the switch to loiter should enable the geofence, and if a min alt was set, should then climb.
I would hope that switching to loiter doesn’t disable the fence.

ChrisBird, 2019-01-14:
@auturgy  Thats what i like about this is that I'm always learning.  Will have to give that a go.

Naterater, 2021-07-26:
I'll chalk this up to possibly available as a feature using geofence.

closed 2021-07-26 by Naterater