ArduPilot/ardupilot #10227: No ESC PWM Signal on PIXHAWK4

Erki312, 2019-01-11:
### Bug report

**Issue details**

I have setup a holybro Pixhawk4 with arducopter Firmware.
I can configure the copter with Mission-planner.
But I have no pwm-signals at the ESCs

I can start the motor-test, but the ESCs don’t get any pwm Signal.
I use this Firmware:

  ArduCopter V3.6.5-rc1 (d8752f0c)
  ChibiOS: d2030d88
  fmuv5 0037001E 3038510A 35323635

I have also testet with 3.7.0 and some other 3.6. Allways the some issue.
I hope for some help.
Regards.
Torsten
### Bug report

**Version**
  ArduCopter V3.6.5-rc1 (d8752f0c)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Holybro Pixhawk4 

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2019-01-11:
please re-flash with[ new firmware for pixhawk4](http://firmware.ardupilot.org/Copter/stable/Pixhawk4/), you will have to use load custom firmware in MP as it is detected incorrectly

auturgy, 2019-01-11:
We need more information to help, but ArduPilot works with Pixhawk4 so this isn’t demonstrably a bug, it’s a user support request. As such it should be in the forum (discuss.ardupilot.org) so I’m closing the issue here.
Please post to the forum, with a description of your setup (which IO you’re using, which ESC’s) and your parameter file.

closed 2019-01-11 by auturgy