ArduPilot/ardupilot #10234: Failsafe and multi-battery missions

skypuppy, 2019-01-13:
### Feature request

**Describe the solution you'd like**

Some of the missions I want to do will require several battery changes. So, is there a failsafe method that 1) allows the drone to continuously calculate the rtl power expenditure versus it's current position, 2) remembers it's last mission waypoint, 3) RTL's to change batteries, and 4) return to it's last waypoint then continues the mission (until the next battery change?) If the RTL expenditure calcs are not continually performed now, how can I make it an official change request, whether there is a mission or not? The goal is to not have the drone run out of juice in the middle of nowhere, making it rather difficult to find and/or retrieve. This especially applies to new birds where there's not much track record of their performance and power loads.

**Describe alternatives you've considered**
There really isn't a good one that I could think of.

If you guys could suggest a section of code to put it in, maybe I could take a look at it.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[ x ] Rover
[ x ] Submarine

rmackay9, 2019-01-14:
(1) is already on the to-do list somewhere and (4) might be as well

IamPete1, 2021-07-26:
We have a lua script example to RTL calculated expected current draw for plane anyway, it works quite well if you have a good wind estimate. 

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/plane-wind-fs.lua

So long as you don't reboot auto will continue where it left off.