ArduPilot/ardupilot #10255: copter triggered RTL while in AUTO mode with failsafe set to 2 (continue mission in auto)

maciek01, 2019-01-16:
### Bug report

**Issue details**

_Please describe the problem_

At 06:33 copter switched to RTL mode while in AUTO mode previously, despite of the following settings:

FS_THR_ENABLE, 2
BATT_FS_LOW_ACT, 0

RCIN channels where at their low positions as shown in the log.
CH7 is mapped to RTL and was always set to low as log indicates.

in order to deactivate RTL and resume AUTO mission i switched  the CH7 to high and back to low twice (just in case).

Independent of this issue at 7:29 when i reactivated AUTO using CH5 copter resumed the mission at WRONG altitude. mission alt was set to 50m rel, copter resumed mission at ~22m.

**Version**
_What version was the issue encountered with_

ArduCopter V3.6.4 (6c4da9e3)

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

Quad X

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

Cube

**Logs**
_Please provide a link to any relevant logs that show the issue_

[00000009.log.txt](https://github.com/ArduPilot/ardupilot/files/2762414/00000009.log.txt)

maciek01, 2019-01-16:
i noticed this in the log

MSG, 238562403, Failsafe: Terrain data missing
ERR, 238562751, 23, 1
MODE, 238562825, RTL, 6, 11
SRTL, 238562855, 1, 10, 150, 0, 16.20799, -26.52515, -49.77391
ERR, 238562873, 23, 0

its a bug in my view. mission intent is to continue no matter what but yet this situation causes diversion in the plan.
WPs are in the end set to absolute values as the planner ought to recalculate

rmackay9, 2019-01-16:
OK, so it's a [terrain failsafe which is documented here](http://ardupilot.org/copter/docs/terrain-following.html#failsafe-in-case-of-no-terrain-data) but perhaps we should also add it to our failsafe's wiki page so I've added this to our [wiki-to-do-list](https://github.com/ArduPilot/ardupilot_wiki/issues/1579).

In this case it's triggered because the range finder goes out of range when the vehicle reaches about 17m.

Playing with semantics perhaps but I wouldn't say it's a "bug" because it is operating as designed, we simply don't support mixing the terrain database with rangefinder data.  [I've added it to the to-do list here](https://github.com/ArduPilot/ardupilot/issues/10257).

closed 2019-01-16 by rmackay9

maciek01, 2019-01-16:
@rmackay9 thank you understood.
I know this belongs in forum now, but would simply defining mission in ABS alt, eliminate the reliance on terrain data during the mission execution? My objective is to deactivate failsafes that interfere with mission.
For one-way missions where LAND location is close to the battery life, RTL is simply not an option.
It would be better strategy to head to the designated LAND position once you cross half way mission mark.