ArduPilot/ardupilot #10258: Copter: TradHeli - RSC not responding in master

IamPete1, 2019-01-16:
Trad Heli's RSC does not respond on Master testing in SITL. I guess it is probably something to do with
https://github.com/ArduPilot/ardupilot/pull/10212 
I will see if I can get to the bottom of it this evening but opened a issue in case someone with more Heli experience could put there finger on the problem straight away.

### Bug report

**Issue details**
no response from heli RSC output. 

**Version**
Master

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

https://github.com/ArduPilot/SITL_Models/tree/master/RealFlight/Tridge/Helicopters/Raptor720

**Hardware type**
realflight SITL

**Logs**

peterbarker, 2019-01-16:
@IamPete1 If you can give steps in SITL to reproduce I can convert to an autotest.  Or you could add one yourself here: https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/arducopter.py#L2730

IamPete1, 2019-01-16:
@peterbarker Thanks for taking a look, its my first play with heli so not %100 sure what the expected behaviour should be, but basically I armed and the RSC stayed at min. I think it should spool up to a given value form the collective - thrust curve. What ever I tried could not get RSC to move off min, in the curve mode or the set-point or the pass-through modes.

rmackay9, 2019-01-16:
@IamPete1, it might just be the interlock switch.  I flew heli after the change went in but here's what's required I think:

1. rc 8 1000 (i.e. disable motor)
2. arm throttle
3. rc 8 20000 (i.e. enable motors)
4. rc 3 2000 (might need to wait a while for the main rotor to spool up)

After this it should fly

peterbarker, 2019-01-16:
@IamPete1 are you using default parameters here?

You might like to read through fly_avc_mission - if that is passing CI then things must generally be working.

You can run it with

`./Tools/autotest/autotest.py --gdb --debug build.Helicopter fly.CopterAVC.AVCMission` (use current master)

We don't go out of our way to test the various rsc modes.

IamPete1, 2019-01-16:
@rmackay9  Thanks, it could just be this, On Tridge's model it uses a throttle curve so it never prompted me to switch the interlock for arming, I think there may be some confusion about what channel the interlock is, setting it using the RC_OPTION to anything other than CH8 maybe is not the correct thing to do? @kd0aij also reported RSC not working having set the interlock and armed on the Quad Heli realflight model. 

@peterbarker I was just using Tridge's params for his model so the default's should be fine, As you say if RSC is not working it shouldn't pass the avc auto test, I will try it this evening, Thanks

IamPete1, 2019-01-16:
@peterbarker  I managed to run the Autotest with some help from Francisco. Using Cygwin I got a divide by zero error on line 362 of common.py - self.progress("Drained %u messages from mav (%f/s)" % (count, count/float(tdelta)))
Just commenting out this line allowed it work fine, not sure If this is a general issue with autotest on Cygwin or just for me. it might be worth testing for the divide by zero? (it passed the auto test fine btw, thanks for adding it)

@rmackay9 Your right it is just the interlock, following your instructions I got it working. I think my misunderstanding was thinking that the heli interlock is the same as the copter one. It seems the heli code ignores any interlocks set using the RC_OPTION and just uses ch8. I assumed the interlock had to be set as with RC_OPTION  as for copter. 

Maybe a new feature request would be to use RC_OPTION for heli interlock? we could still default it to be ch8 I guess. I will look at doing a PR once I have this monocopter sorted.

closed 2019-01-16 by IamPete1

rmackay9, 2019-01-16:
@IamPete1, yes we should certainly be using the RCx_OPTION to allow moving the heli interlock around.  I'm actually surprised that's not how it works.  I think that by default it is channel 8 but the user can use RCx_OPTION to assign it to another channel.  @bnsgeyer is this correct?

bnsgeyer, 2019-01-17:
@rmackay9.  After looking through the code, I don’t believe we can use the RCX option for interlock enable/disable.  Part of the issue with this is for one of the motor speed control options which is pass-through, it requires the RC channel PWM value to set the rotor speed. I will look at this in the coming week to see if I can put together a PR to fix this.