ArduPilot/ardupilot #10270: Uncontrolled Yaw

mwingerson, 2019-01-17:
### Bug report

**Issue details**
I'm building a custom quadplane and I'm seeing a an uncontrollable yaw. As soon as the bird takes off, it spins in yaw with no input from the receiver. First, I thought it was magnetometer but a better placement didn't help. Then I thought yaw P was low and we got it flying well with some tuning. We took it out the next day and it spun in yaw again. To further clarify, the RC is calibrated and the yaw stick is centered and oriented correctly.

The one thing that I can't get my head around is why the desired yaw and observed yaw look the way they do. Also in this log, I noticed a differential in motors that could indicate fighting or movement in yaw.

Thanks for the help!
Marshall

**Version**
ArduPlane V3.10.0-dev (7430cd3c)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane - quadplane
[  ] Rover
[  ] Submarine

**Airframe type**
Custom octocoax quadplane

**Hardware type**
Pixhawk 2.1 with Here+ RTK kit

**Logs**
https://plot.dron.ee/QAog

IamPete1, 2019-01-17:
Check your motor and props are rotating in the correct direction.

rmackay9, 2019-01-18:
It's possible [this fix that just went into master](https://github.com/ArduPilot/ardupilot/pull/10232) could be the cause.  The good news would be that a future release will resolve the issue.

lthall, 2019-01-18:
This is a setup issue with your aircraft. You may have your props on backwards or flex in you wing is overpowering the yaw authority of your props.

This is not associated with the EKF yaw reset fix that Randy mentions. That fix prevents a small step change in yaw heading on EKF reset.

closed 2019-01-18 by lthall

mwingerson, 2019-01-30:
@lthall I'm not convinced it is a problem with the setup. I have a very similar build that is doing fine and I have gone through enough of the components that I am fairly confident the components are good. Normally, I would agree with you except for the inconsistent nature of it. When I have had a setup issue in the past, it has all been a consistent issue. 

Here is a log where the bird operates fine for a couple minutes in hover before having issues in yaw as it lands. How can that be a setup issue? I am purely asking since I don't understand your perspective.

https://plot.dron.ee/3Yhw

rmackay9, 2019-01-30:
@mwingerson,

I tend to agree with @lthall.  Checkout the RC outputs.  There is a consistent offset between the clockwise vs counter-clockwise propellers which indicates some kind of setup or calibration issue with the motors or ESCs.
[image]

lthall, 2019-01-30:
Can you provide a video and the log that goes with that video so we can see what is happening?

mwingerson, 2019-01-30:
@lthall 
Sorry, I have been using Dronee Plotter for awhile and forgot other people don't use it. You can download the log from the download button in the top right :)

Here is a link to the log: [link](https://drive.google.com/file/d/1FJ1Ec6LRicgvdRi8YGxKUbyxxjRmxAUR/view?usp=sharing)

Unfortunately I don't have any footage of this flight. I'm getting some more testing done today.

@rmackay9 
I have an octocoax with different props on top and bottom so two different groupings of PWMs is expected. 

[image]

rmackay9, 2019-01-30:
@mwingerson, I think though, that regardless of the reason, the large motor imbalance between some props and others is going to limit the vehicle's ability to control itself well because during some maneuvers all those bottoms with high PWM will soon run out of range.  The vehicle will fly more reliably if all the motor outputs can be brought closer together.

JBB2068, 2019-06-06:
> ### Bug report
> **Issue details**
> I'm building a custom quadplane and I'm seeing a an uncontrollable yaw. As soon as the bird takes off, it spins in yaw with no input from the receiver. First, I thought it was magnetometer but a better placement didn't help. Then I thought yaw P was low and we got it flying well with some tuning. We took it out the next day and it spun in yaw again. To further clarify, the RC is calibrated and the yaw stick is centered and oriented correctly.
> 
> The one thing that I can't get my head around is why the desired yaw and observed yaw look the way they do. Also in this log, I noticed a differential in motors that could indicate fighting or movement in yaw.
> 
> Thanks for the help!
> Marshall
> 
> **Version**
> ArduPlane V3.10.0-dev ([7430cd3](https://github.com/ArduPilot/ardupilot/commit/7430cd3c1da011f0cdd2b86c426cd48f4ea7937e))
> 
> **Platform**
> [ ] All
> [ ] AntennaTracker
> [ ] Copter
> [X] Plane - quadplane
> [ ] Rover
> [ ] Submarine
> 
> **Airframe type**
> Custom octocoax quadplane
> 
> **Hardware type**
> Pixhawk 2.1 with Here+ RTK kit
> 
> **Logs**
> https://plot.dron.ee/QAog

Did you ever find what was causing this? I am having the same problem with Quadplane on 3.10 Dev.  It yaws on its own and I cannot correct it.

rmackay9, 2019-06-06:
@JBB2068, I think in the case above it was a motor imbalance.  I had another quick look and it seems very clear to me that motor1 is hitting it's upper limit

mwingerson, 2019-06-06:
@JBB2068, I have left the project but I was confident it was an issue with
the airframe not holding the motors square. would you post a log?

JBB2068, 2019-06-06:
I took some more time to level the motor mounts and I believe that is the problem. I was able to get a more controlled flight after eyeballing them on a flat surface. I am going to do more to align them.