ArduPilot/ardupilot #10279: Undefined name 'autotest' in ./Tools/autotest/sim_vehicle.py

cclauss, 2019-01-18:
Discovered via #10278

### Bug report

[flake8](http://flake8.pycqa.org) testing of https://github.com/ArduPilot/ardupilot on Python 3.7.1

$ __flake8 . --count --select=E901,E999,F821,F822,F823 --show-source --statistics__
```
./Tools/autotest/sim_vehicle.py:396:39: F821 undefined name 'autotest'
    swarminit_filepath = os.path.join(autotest, "swarminit.txt")
                                      ^
1     F821 undefined name 'autotest'
1
```
__E901,E999,F821,F822,F823__ are the "_showstopper_" [flake8](http://flake8.pycqa.org) issues that can halt the runtime with a SyntaxError, NameError, etc. Most other flake8 issues are merely "style violations" -- useful for readability but they do not effect runtime safety.
* F821: undefined name `name`
* F822: undefined name `name` in `__all__`
* F823: local variable name referenced before assignment
* E901: SyntaxError or IndentationError
* E999: SyntaxError -- failed to compile a file into an Abstract Syntax Tree

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

peterbarker, 2019-01-18:
Pushed a fix in for this.

sim_vehicle.py is compliant with `flake8 sim_vehicle.py` now - should we tighten things up in any way, or just make sure we run things in CI?

cclauss, 2019-01-18:
Nice fix!  Thanks!  #10278 finds issues in /modules including UAVCAN/pyuavcan#51 that was just fixed as well.  I think for this repo the CI test is good enough for now.

closed 2019-01-18 by cclauss