ArduPilot/ardupilot #10293: Feature request-add osd power display

sheltercv, 2019-01-21:
**Feature request**
add osd power display. benefit estimate loiter flight time and motor work efficiency.

**Platform**
[  ] All
[  ] AntennaTracker
[√] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Matek F405

**firmware type**
Chibios

SergeyBokhantsev, 2019-01-21:
Good request, also it will be great to have an option to show either an instant power value or an averaged for last N seconds.

Rusty105, 2019-04-23:
@SergeyBokhantsev  doesn't the existing "current display" do this already?

Rusty105, 2019-04-23:
@sheltercv  Only while in loiter mode, or continuously displayed?
Something like mm.ss till battery warning?

Rusty

IamPete1, 2021-08-15:
We have horizontal speed efficiency and vertical speed efficiency, battery current draw, battery percentage and flight time.