ArduPilot/ardupilot #10295: ArduCopter autotests fail if --frame octa-quad (ATC_RLL_D gets zeroed)

amilcarlucas, 2019-01-21:
### Bug report

ArduCopter autotests for octa-quad-frame fail

The cause seams to be that ATC_RAT_RLL_D is set to 0, making the pre-arm checks fail
that has to be a bug because the octa-quad parameter file sets that one to 0.022 and the SITL gets that message.

to reproduce this issue:

/Tools/autotest/autotest.py build.ArduCopter defaults.ArduCopter fly.Arducopter --frame octa-quad

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[ *] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
octa-quad

**Hardware type**
SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2019-01-21:
Probably due to the this https://github.com/ArduPilot/ardupilot/pull/10214

richardor-cell, 2020-02-09:
Hi, please where could be found X8 sdf example file as for the iris quad demo to reproduce this ?

IamPete1, 2021-08-29:
Fixed by https://github.com/ArduPilot/ardupilot/pull/10214 

We now allow 0 d term, but not negative.

closed 2021-08-29 by IamPete1