ArduPilot/ardupilot #10296: JSBSim control_port not changing according to instance (SITL)

fabioandraderj, 2019-01-21:
### Bug report

**Issue details**

The control_port of JSBSim is not being changed according to the instance at line 63 of file SIM_JSBSim.cpp. The instance is always 0 even though I start ardupilot with "-I 1" or "-I 2", etc.

Also, it seems that the JSBSim process is not being killed when the ardupilot simulation is closed. Therefore, I have to kill the process manually to be able to open a new simulation.

**Version**
ArduPlane V3.10.0-dev

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

**Hardware type**
SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_

fabioandraderj, 2019-01-21:
I noticed that my previous ardupilot version had the control_port being defined in the function "create_templates". 

`bool JSBSim::create_templates(void)
{
    if (created_templates) {
        return true;
    }
    control_port = 5505 + instance*10;
    fdm_port = 5504 + instance*10;

    asprintf(&jsbsim_script, "%s/jsbsim_start_%u.xml", autotest_dir, instance);
    asprintf(&jsbsim_fgout,  "%s/jsbsim_fgout_%u.xml", autotest_dir, instance);`

IamPete1, 2021-08-15:
@fabioandraderj Is this still a issue?

fabioandraderj, 2021-08-16:
Regarding the instances issue, I cannot say for sure if it was fixed (I didn't test) but from what I could check in the code, nothing was changed, so it should still be an issue.

To fix that, I guess that you should move the lines 63, 64, 67 and 68 to line 79 in the SIM_JSBSim.cpp file

Regarding the issue with the process JSBSim not being killed when Ardupilot is closed is fixed, I don't know if it was fixed.