ArduPilot/ardupilot #10298: Setup signing works over mavlink 1 when it is not supposed to

mnihalm, 2019-01-21:
### Bug report

**Issue details**
MAVLINK 2  - setup signing, used to authenticate GCS is supposed to be a mavlink 2 message.  This by definition is therefore not supposed to work on a link that is set to MAVLINK 1 protocol.

How This was tested - 
- Set Serial port 2 to mavlink 1 at 57600 baud. 
- Connect a 3DR radio to the TELEM2 port (serial 2)
- Connect a second 3DR radio to a computer and run mavproxy using the command `mavproxy.py --master=/dev/ttyUSB0,57600 --console`
- connect a third 3DR radio to a different computer and run the same command to connect to the pixhawk
- both GCSs' can now send commands  to the pixhawk.
- have one GCS setup signing using the command `signing setup passphrase`
- the other GCS can now no longer send commands until the signing key is setup using `signing key passphrase`

I'm not sure if I'm missing something here, but this is scary that someone can just connect to a pixhawk and then hijack the connection, all without me being unable to do anything about it. This seems like a easier attack than the older method discovered years ago where the net_id of the radio was scanned and spoofed. (I think. Can't recall correctly)

I'm not sure if I'm doing something wrong, but I've attached the param file for reference along with the telemetry logs from both the pcs (which happened to be raspberry pi's). 
[mavlink signing test.zip](https://github.com/ArduPilot/ardupilot/files/2780121/mavlink.signing.test.zip)

**Version**
Plane 3.9.4
Might be a cross platform issue. Haven't tested
**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
None, pixhawk and 3dr radios connected to two computers
**Hardware type**
pixhawk and 3dr radios
**Logs**
_Please provide a link to any relevant logs that show the issue_

khancyr, 2019-01-21:
Hello,
I am pretty sure it is normal. When ardupilot received v2 message it switch to v2. And mavproxy is v2 by default. So you where never in v1 but always in v2...
And there is not hijack as without the key, you cannot do anything

auturgy, 2019-01-21:
Pierre if this is reproducible (sounds like it is), the issue here seems to be that if signing isn’t set up, someone can setup the key / initiate signing mid flight - potentially hijacking the vehicle.
If that is in fact the case it should be fixed. Rejecting a key if armed would be enough to prevent it.
@tridge

mnihalm, 2019-01-22:
Pierre, I didn't know ardupilot worked that way. If what you say is tru then the behavior is not abnormal. But as @auturgy said, the issue is for drones that do not have a key setup, someone can easily add a key and then hijack the drone.

mnihalm, 2019-01-22:
Ah, Of course. sniffing the netID is required like you mentioned. 
One concern for the same is that the default settings in all Sik and RFD radios have the same netID of 25. Perhaps we should add a warning to let users change the netID in the GCS to prevent this.

Any beginner user would not change the settings in the radio and would end up having the same default netID , where things can go wrong as discussed. 

Best,
Nihal

OXINARF, 2019-01-23:
@mnihalm I'm closing this as by design.

If I got it correctly, your main complaint is that, when setting a port to talk MAVLink1, it still understands incoming MAVLink2 messages. This is by design (of the official parser generator), setting a port MAVLink protocol only says what it outputs, not what the receiving end understands.

Also see that, by design, when enabling MAVLink2, the autopilot starts by outputting MAVLink1 and only starts outputting MAVLink2 if it receives a MAVLink2 message (an exception is if signing is already enabled, in that case, it will send MAVLink2 from the start).

#10311 done by @peterbarker reduces the attack vector by only allowing signing to be setup while disarmed. But realize that MAVLink1 isn't secure and was never intended to be. MAVLink2 isn't either unless after you setup signing **over a secure communication channel**.

closed 2019-01-23 by OXINARF

mnihalm, 2019-01-24:
Ah, Thanks for the inputs !
I had definetly not understood the the design behind the protocols accepting and receiving MAVLINK messages. 
Thanks to peterbarker for fixing it quickly.

While the fix reduces the attack vector as you mentioned, this vulnerability would be a very easy way to attack. sniffing netID takes some effort, whereas just connecting to the radio and sending a message is too easy an attack, and I wanted to get it noticed.  I'm glad it's fixed now.

Again, thanks for everything 😄