ArduPilot/ardupilot #10302: Quadplane: pitch rudder to operate in VTOL modes

najia2010, 2019-01-22:
@tridge[https://github.com/ArduPilot/ardupilot/issues/4459](url)
The latest firmware 3.9.5 and V3.10.0-dev VTOL mode, the fixed-wing pitch rudder can not work when starting to take off, the windy weather is very unfavorable for the aircraft to stabilize. Request to add a switch to control the pitch rudder. Losing the fixed-wing rudder surface self-stabilization in bad weather is really bad for VTOL mode. Thank you

magicrub, 2019-01-22:
closing, duplicate of https://github.com/ArduPilot/ardupilot/issues/4459

closed 2019-01-22 by magicrub

najia2010, 2019-01-23:
@magicrub:I very much hope that in VTOL mode, the fixed-wing pitch rudder can help the aircraft control the flight attitude during the take-off phase.。Do you have any good suggestions?

magicrub, 2019-01-23:
seems like a useful fix, but posting a duplicate thread makes us spend time explaining that it's a duplicate thread instead of actually working on stuff. I'm freezing msgs on this thread, please use the other one