ArduPilot/ardupilot #10305: Airspeed: connecting two SDP3X not working

aero4, 2019-01-22:
### Bug report

**Issue details**

I have a self-made SDP31 board with two SDP31 and each has its own I2C address (accomplished with the external resistor, the first sensor has the address 0x22, the second 0x23). Both sensors are working independently on I2C bus 0.

The problem is, that when ARSPD_TYPE and ARSPD2_TYPE are both set to “6” (SDP3X), then the first SDP31 is initialized two times (with address 0x22) and the logged value for ARSPD1 and 2 are the same.

When I uncomment the continue instruction from the SDP3X initialization (inside the "stop continuous average mode" if-instruction; line 77 of AP_Airspeed_SDP3X.cpp) in the source code, both sensors (with addresses 0x22 and 0x23) are recognized but only the second sensor is logged (first sensor only shows 0). 
So it seems that only one sensor is supported of the same type and the second one is overwritten.

**Version**
3.9

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Fixed wing

**Hardware type**
Pixhawk 2 Cube

I am also interested in a temporary fix, because I need two airspeed sensors to be logged as fast as possible.

Thank you!

magicrub, 2019-01-22:
@aero4 I took a glance at the [data sheet](https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/0_Datasheets/Differential_Pressure/Sensirion_Differential_Pressure_Sensors_SDP3x_Digital_Datasheet.pdf) and the driver and see that the addresses are 0x21, 0x22, 0x23. Have you tried using the first two instead of the last two? It shouldn't matter but it's worth a try.

aero4, 2019-01-22:
I bypassed the ground resistor and then UART says that two SDP31 Sensors with the address 0x21 were found...

magicrub, 2019-01-22:
@aero4 so one device's addr pin (pin 9) was 0 ohm to grounded and on another device's pin was 1.2k ohm to ground and that resulted in two devices with address 0x21?
Can you post the schematic?

aero4, 2019-01-22:
Yes, this is exactly what I did.
Schematic:
https://ibb.co/31QxCzw

magicrub, 2019-01-22:
also, for the record, you should remove the 10k pullup. Pull-ups are always on the master.

aero4, 2019-01-22:
Unfortunatelly my SDP31 board does not allow different buses, is there a temporary fix which I can make?

As said beneath the picture R3 and R4 are left bank but thank you for the hint

WickedShell, 2019-01-22:
The only temporary one I know of would be to make a patch like https://github.com/ArduPilot/ardupilot/pull/7313 which allowed the MS4525 address to be specified. (Or just add a `ARSPD_ADDR` param and if it's not 0 use that instead.

aero4, 2019-01-22:
I found a temporary solution for me:
I created a new sensor called SDP3Xmult and assigned it to ARSP_TYPE 9.
In the SDP3Xmult.cpp file I set all addresses to 0x00 except of 0x23.
-> now i can log both sensors!

If anyone else has this problem here is a more detailed description:

- add files _AP_Airspeed_SDP3Xmult.cpp_ and _AP_Airspeed_SDP3Xmult.h_ to the _libraries/AP_Airspeed_ directory
- extend enum _airspeed_type_ in file _AP_Airspeed.h_ by one sensor with the number _9_: _TYPE_I2C_SDP3XMULT=9_
- include file _AP_Airspeed_SDP3Xmult.h_ into the file _AP_Airspeed.cpp_: _#include "AP_Airspeed_SDP3Xmult.h"_
- create new intry into the switch case with the denotation _switch ((enum airspeed_type)param[i].type.get())_:
_case TYPE_I2C_SDP3XMULT:
            sensor[i] = new AP_Airspeed_SDP3Xmult(*this, i);
            break;_
- set EEPROM parameter _ARSPD_BUS_ und _ARSPD2_BUS_ to _0_
- set EEPROM parameter _ARSPD_TYPE_ to _6_ and _ARSPD2_TYPE_ to _9_

Here are my _SDP3Xmult_ files (ending _.txt_ has to be removed):
[AP_Airspeed_SDP3Xmult.h.txt](https://github.com/ArduPilot/ardupilot/files/2784888/AP_Airspeed_SDP3Xmult.h.txt)
[AP_Airspeed_SDP3Xmult.cpp.txt](https://github.com/ArduPilot/ardupilot/files/2784889/AP_Airspeed_SDP3Xmult.cpp.txt)

Evan1010, 2019-01-24:
This seems similar to a [user problem on Discuss](https://discuss.ardupilot.org/t/dual-airspeed-sensor-problem/34131/13) a while back, where he could not get two 4525DSO sensors to read separately even when on separate I2C busses.

This could be problematic, if two sensors of the same type number cannot be used - I and a couple others were planning to test with dual UAVCAN airspeed sensors.

WickedShell, 2019-01-24:
@Evan1010 yeah MS4525 will always scan all buses in the same order as I described on the other issue.

Evan1010, 2019-01-24:
@WickedShell So am I understanding correctly that - for the MS4525 driver - the I2C Bus parameter has no effect? This is what we had theorized with the afore linked Discuss topic. 

For the other issue, are you referring to Pull Request [#7313](https://github.com/ArduPilot/ardupilot/pull/7313)?
If I understand correctly that it scans the buses 0 to n and grabs the first MS4525 it sees regardless of the bus parameter, I can see how that happens.

aero4, 2019-01-24:
Another problem, which is the main problem for the bug described here, is, that if a ARSP1 sensor is found at the address e.g. 0x22, this address is not flagged as occupied and therefore the same address (and sensor) is used for ARSP2...