ArduPilot/ardupilot #10306: SDP3X uses bus type 0 (when connected to I2C 2), all other sensors bus type 0

aero4, 2019-01-22:
### Bug report

When I connect a MS4525D to I2C 2 port of my Pixhawk 2 Cube I have to set ARSPD_BUS to 2 (auxillary).
But when using a SDP31 I have to set the ARSPD_BUS value to 0 (internal). Otherwise the sensor is not recognized.
I don't think this behaviour is intended?

**Version**
3.9

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Fixed wing

**Hardware type**
Pixhawk 2 Cube

WickedShell, 2019-01-22:
The MS4525 driver scans all I2C buses and ignores the `ARSPD_BUS` parameter. I attempted to push through a change so it respected it, but it was shot down as it would break anyone who was currently relying upon that behavior.

With that being said I don't think we will be changing any of the other sensors to automatically scan all the buses so this is the intended behavior as it causes problems when attempting to mix and match sensors. (For example you can't use a MS4525 and a DLVR sensor at the moment unless the MS4525 is on the lowest numbered bus, which has caused me problems).

closed 2019-01-22 by WickedShell

aero4, 2019-01-22:
But when I left ARSPD_BUS at 0 my SDP31 was not recognized

WickedShell, 2019-01-22:
ARSPD_BUS 0 doesn't actually mean scan all the buses, it means use the internal bus. The MS4525 driver however ignores the parameters and always scans the buses, you could have any value in that parameter and the MS4525 would initialize in the same way. (And for this reason you can't actually do dual MS4525's either)

aero4, 2019-01-22:
But if the MS4525 scans all buses and SDP31 driver not, where is that behavior documented?
It costed me many hours to find out that i have to set the bus to 0 istead of the standard value 2 and to understand it was not a hardware fault...

WickedShell, 2019-01-22:
It's not... I think the MS4525 behavior is a bug since we have this parameter sitting next to it that doesn't behave as expected. Unfortunately I've been unable to convince @tridge to let the ARSPD_BUS parameter work correctly for the 4525.

(I've spent many an hour trying to solve the same set of problems, an amazing amount of extra time was spent on chasing what happens with autocal turned off and an ARSPD_OFFSET of 0.... (hint: we consider it unhealthy and refuse to use it)).

aero4, 2019-01-22:
Ok I understand, it seems that the Airspeed feature is a little bit buggy yet... I hope I can get it run somhow...