ArduPilot/ardupilot #10320: Battery Failsafe causes RTL during LAND

Naterater, 2019-01-23:
### Bug report

**Issue details**

Battery Failsafes can be triggered during a LAND mission item. I see no case where this is a good idea, and it will often cause crashes due to the increased power requirements and low battery capacity at the end of a mission.

I see this happening often where the BATT_FS_LOW_ACT = 1 (RTL) and the BATT_FS_CRT_ACT = 4 (QLAND). During the land, a low battery failsafe may be engaged causing an exit of the current landing and a transition to forward flight. That's very power-hungry, and I've heard of complaints. I reproduced in SITL, parameters below. It's much better to just continue the landing during any battery failsafe.

**Version**
Plane 3.10 dev

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All - Tested in SITL by setting the following parameters:
BATT_CRT_VOLT = 11
BATT_FS_CRT_ACT = 1 (RTL)
then command a QLAND followed by
SIM_BATT_VOLTAGE = 10

**Hardware type**
SITL

**Logs**
N/A

Naterater, 2019-01-23:
I noticed your case, and thanks for linking it here. Technically this is a separate failsafe action and scenario, but I agree this should change.

magicrub, 2019-01-23:
That was supposed to be handled in [this code](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/events.cpp#L153) but it seems the logic is wrong.

Naterater, 2019-04-01:
I’m reopening this until #10321 makes its way to the stable releases, and this should have been done by now. This behavior is bad enough that it was mentioned that it should be back-ported in the PR.

@wickedshell @tridge @magicrub

ovolkmann, 2019-04-01:
This is a pretty serious issue especially for QuadPlanes. We've almost lost a few birds as a result of battery failsafe taking place during a land sequence during an auto mission where the landing spot was different from the RTL location. As you may be aware, quadplanes suck a lot of energy during transition to copter mode, take-off and landing. When the battery failsafe kicks in during landing sequence at a location other than home, the quadplane attempts RTL, sometimes from 1m above the ground...

Naterater, 2019-04-02:
As discussed on the dev call, the merged commit was indeed back-ported as of January 24th; it was just missed in the release notes.

Naterater, 2019-09-20:
I'm reopening this as it is actually still an issue that is now duplicated here: https://github.com/ArduPilot/ardupilot/issues/12264

Pedals2Paddles, 2019-09-21:
This is still an issue in copter. If you're in land mode, and the battery or radio failsafe kicks in set for RTL, it will abort the landing and execute a complete RTL.  I tried to make it be as simple as just don't do that since you're already landing. Or a simple parameter to enable/disable this behavior.  But one person insisted that was very bad because he uses land mode when not landing...  Instead, it was desired to make an extremely complicated multi-system set of bitmasks to accomplish this one simple and logical thing. 

I started tinkering with making this overly complicated series of bitmasks.  Other changes took place which completely hosed everything I had done so far (it happens, nobody's fault). But I just ran out of time to keep making something that should be simple overly complicated over and over.  It is not done and to my knowledge, there isn't anyone else working on it either.

proficnc, 2019-09-21:
Land mode is for landing... anyone using it for “alternative” reasons is not a justification for causing this problem. 

Did you do a PR for your solution Matt?

rmackay9, 2019-09-21:
So as not to mix up requests, I've clarified that this request from @Naterater is for Plane.

Naterater, 2019-09-21:
https://github.com/ArduPilot/ardupilot/issues/6390 is the request for copter. There are multiple references to this same issue now spread throughout github and discuss... and all discussion points to the bad and uxpected behavior of failsafes during landing. I'm looking forward to having these issues closed.

Pedals2Paddles, 2019-09-21:
My annoyance over this whole thing prompted me to update my PR for the copter one....

IamPete1, 2020-02-18:
closed by #13065 for basic missions this should now be fixed, if you have complex missions where there are jumps after the do land start you will need to enable distance based calculation with MIS_OPTIONS bit 1, ie + 2 to what you have set

closed 2020-02-18 by IamPete1