ArduPilot/ardupilot #10326: seems to have found a bug in firmware 3.9.4 that is very much expected to be fixed.

najia2010, 2019-01-24:
updating the 3.9.4 firmware, all fixed-wing rudder surfaces will not participate in self-stabilization in vtol mode as long as the GPS is active and unlocked，How to set the fixed-wing rudder surface to participate in the self-stabilization of the fuselage? Mine is the delta wing VTOL Y3 structure. At high altitude QLOITER, it is very unstable due to the wind. It requires a fixed-wing rudder surface to participate in the stability of the fuselage. Expectations can be fixed in 3.9.5.

auturgy, 2019-01-24:
Duplicate of https://github.com/ArduPilot/ardupilot/issues/4459
@najia2010  we have provided feedback to you a number of times linking the issues you open regarding this problem to the pre-existing one already in the system.  We try to not have duplicates of already reported issues, so I'm closing this and linking back to 4459 (as per the last couple of times you've raised this issue).

closed 2019-01-24 by auturgy

najia2010, 2019-01-24:
@auturgy  ：I don't seem to understand the rules, sorry.
I found this error after updating the firmware. I hope that the firmware can be fixed in subsequent firmware. Yesterday, during the vtol test, I found that as long as the GPS signal was shielded by the foil, the rudder surface of the fixed wing would participate in self-stabilization. However, once the GPS signal is obtained, the fixed-wing rudder surface will lose its self-stabilizing function. Since I am a vtol enthusiast, the system construction principle is still unclear. I don't know how to express the problem according to the requirements of the forum, so I can only try it and describe this phenomenon.
What should I do now?

auturgy, 2019-01-24:
Thanks for reporting.  It sounds like you need some help with setting up your vtol aircraft: posting to the forum in this category: https://discuss.ardupilot.org/c/arduplane/vtol-plane will make it most visible to those who can help.  There is also some information on the vtol weathervane function here: http://ardupilot.org/plane/docs/quadplane-weathervaning.html
As above, the specific issue with rudder mixing in VTOL is already documented.
Thanks again, and good luck!

najia2010, 2019-02-01:
@auturgy Thank you very much for your help, the forum you mentioned, because of network problems, unable to speak (no VPN). So I can only ask for your help again.
This week I debugged according to the wind vane parameters, but found that the pitch rudder could not participate in controlling the flight attitude. I hope that all the rudder surfaces, including the pitch rudder, the lateral rudder and the rudder, are involved in flight attitude control. Especially the pitch rudder participates in attitude control (my plane is Y3)
Is there a parameter to control the pitch rudder to participate in attitude control at all times?
Thanks again.