ArduPilot/ardupilot #10333: sub: libuavcan dsdl failed to generate with StopIterator

patrickelectric, 2019-01-24:
### Bug report

**Issue details**

Can't build with waf to Pixhawk1.

Libuavcan failed with:
```
libuavcan_dsdl_compiler.DsdlCompilerException: generator raised StopIteration
generator raised StopIteration
```

I'm running python 3.7.2;

**Version**
master - ff52ceabba4505999a5795d56606893eb1e00150

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[x] Submarine

**Hardware type**
Pixhawk1 (Moved from px4-v2)

**Logs**
```sh
 patrick@pichau  /tmp  git clone github:ardupilot/ardupilot
Cloning into 'ardupilot'...
remote: Enumerating objects: 26, done.
remote: Counting objects: 100% (26/26), done.
remote: Compressing objects: 100% (20/20), done.
remote: Total 243851 (delta 9), reused 18 (delta 6), pack-reused 243825
Receiving objects: 100% (243851/243851), 121.22 MiB | 7.93 MiB/s, done.
Resolving deltas: 100% (179057/179057), done.
 patrick@pichau  /tmp  cd ardupilot 
 patrick@pichau  /tmp/ardupilot   master  git submodule update --init --recursive
Submodule 'modules/ChibiOS' (git://github.com/ArduPilot/ChibiOS.git) registered for path 'modules/ChibiOS'
Submodule 'modules/gbenchmark' (git://github.com/google/benchmark.git) registered for path 'modules/gbenchmark'
Submodule 'gtest' (git://github.com/ArduPilot/googletest) registered for path 'modules/gtest'
Submodule 'modules/mavlink' (git://github.com/ArduPilot/mavlink) registered for path 'modules/mavlink'
Submodule 'modules/uavcan' (git://github.com/ArduPilot/uavcan.git) registered for path 'modules/uavcan'
Submodule 'modules/waf' (git://github.com/ArduPilot/waf.git) registered for path 'modules/waf'
Cloning into '/tmp/ardupilot/modules/ChibiOS'...
Cloning into '/tmp/ardupilot/modules/gbenchmark'...
Cloning into '/tmp/ardupilot/modules/waf'...
Cloning into '/tmp/ardupilot/modules/gtest'...
Cloning into '/tmp/ardupilot/modules/uavcan'...
Cloning into '/tmp/ardupilot/modules/mavlink'...
Submodule path 'modules/ChibiOS': checked out '35ad8a79576e793114dd9288fc12c06d93a151d1'
Submodule path 'modules/gbenchmark': checked out '006d23ccca1375a973b7fae0cc351cedb41b812a'
Submodule path 'modules/gtest': checked out 'c99458533a9b4c743ed51537e25989ea55944908'
Submodule path 'modules/mavlink': checked out 'a9da2d2efa6a7d16004d5cde11ce1559aab645db'
Submodule 'pymavlink' (git://github.com/ArduPilot/pymavlink.git) registered for path 'modules/mavlink/pymavlink'
Cloning into '/tmp/ardupilot/modules/mavlink/pymavlink'...
Submodule path 'modules/mavlink/pymavlink': checked out 'c18db673f0214b683bc330587bc11ce2818a8e8d'
Submodule path 'modules/uavcan': checked out '72d4b9aecb5b23a142827bea962f636423d99041'
Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'modules/uavcan/dsdl'
Submodule 'libuavcan/dsdl_compiler/pyuavcan' (https://github.com/UAVCAN/pyuavcan) registered for path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan'
Cloning into '/tmp/ardupilot/modules/uavcan/dsdl'...
Cloning into '/tmp/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/pyuavcan'...
Submodule path 'modules/uavcan/dsdl': checked out '192295c4f9b67f4a20b0eabf74757b6597415f2b'
Submodule path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan': checked out 'c58477a644d20ccf95a20c151f3a0402f271c3b8'
Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl'
Cloning into '/tmp/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl'...
Submodule path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl': checked out 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466'
Submodule path 'modules/waf': checked out '337b0a96beeac517ecde081fb76dedd039089734'
[... 177 lines cut ...]
  File "/tmp/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/pyratemp.py", line 1119, in render
    for i in loop_iter:
RuntimeError: generator raised StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/tmp/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/libuavcan_dsdlc", line 61, in <module>
    dsdlc_run(args.source_dir, args.incdir, args.outdir)
  File "/tmp/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 62, in run
    run_generator(types, output_dir)
  File "/tmp/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 111, in run_generator
    die(ex)
  File "/tmp/ardupilot/modules/uavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 89, in die
    raise DsdlCompilerException(str(text))
libuavcan_dsdl_compiler.DsdlCompilerException: generator raised StopIteration
generator raised StopIteration

uavcangen returned 1 error code
Waf: Leaving directory `/tmp/ardupilot/build/Pixhawk1'
Build failed
 -> task in 'uavcan' failed (exit status 1): 
        {task 139758841556328: uavcangen CoarseOrientation.uavcan,Timestamp.uavcan,1010.ArrayCommand.uavcan,1011.Status.uavcan,Command.uavcan,1000.Solution.uavcan,1001.MagneticFieldStrength.uavcan,1002.MagneticFieldStrength2.uavcan,1003.RawIMU.uavcan,1020.TrueAirspeed.uavcan,1021.IndicatedAirspeed.uavcan,1025.AngleOfAttack.uavcan,1026.Sideslip.uavcan,1027.RawAirData.uavcan,1028.StaticPressure.uavcan,1029.StaticTemperature.uavcan,1040.AngularCommand.uavcan,1041.GEOPOICommand.uavcan,1044.Status.uavcan,Mode.uavcan,1110.Temperature.uavcan,1030.RawCommand.uavcan,1031.RPMCommand.uavcan,1034.Status.uavcan,1060.Fix.uavcan,1061.Auxiliary.uavcan,1062.RTCMStream.uavcan,1063.Fix2.uavcan,ECEFPositionVelocity.uavcan,1070.Command.uavcan,1071.Status.uavcan,1129.FuelTankStatus.uavcan,1120.Status.uavcan,CylinderStatus.uavcan,1080.BeepCommand.uavcan,1081.LightsCommand.uavcan,RGB565.uavcan,SingleLightCommand.uavcan,1090.PrimaryPowerSupplyStatus.uavcan,1091.CircuitStatus.uavcan,1092.BatteryInfo.uavcan,1050.Measurement.uavcan,1100.ArmingStatus.uavcan,2000.GlobalNavigationSolution.uavcan,1.GetNodeInfo.uavcan,2.GetDataTypeInfo.uavcan,341.NodeStatus.uavcan,4.GetTransportStats.uavcan,4.GlobalTimeSync.uavcan,5.Panic.uavcan,5.RestartNode.uavcan,6.AccessCommandShell.uavcan,CANIfaceStats.uavcan,DataTypeKind.uavcan,HardwareVersion.uavcan,SoftwareVersion.uavcan,16370.KeyValue.uavcan,16383.LogMessage.uavcan,LogLevel.uavcan,1.Allocation.uavcan,30.AppendEntries.uavcan,31.RequestVote.uavcan,390.Discovery.uavcan,Entry.uavcan,15.Begin.uavcan,380.Indication.uavcan,40.BeginFirmwareUpdate.uavcan,45.GetInfo.uavcan,46.GetDirectoryEntryInfo.uavcan,47.Delete.uavcan,48.Read.uavcan,49.Write.uavcan,EntryType.uavcan,Error.uavcan,Path.uavcan,10.ExecuteOpcode.uavcan,11.GetSet.uavcan,Empty.uavcan,NumericValue.uavcan,Value.uavcan,2010.Broadcast.uavcan,63.Call.uavcan,Protocol.uavcan -> }
 (run with -v to display more information)
```

patrickelectric, 2019-01-24:
running waf with python2 fixed the problem.

WickedShell, 2019-01-24:
@patrickelectric yeah this is due to [PEP 479](https://www.python.org/dev/peps/pep-0479/) I'd love to see this solved as it's forcing me down the Python2 road as well that you found.

DaHaiHuha, 2019-04-11:
yes, i have come cross the same question, but how to use Python2 to build?

patrickelectric, 2019-04-11:
`./ waf ... -python=python2`

OXINARF, 2019-04-19:
Fixed with #11144.

closed 2019-04-19 by OXINARF