ArduPilot/ardupilot #10346: Copter: SUMD not working 3.6.5

lvale, 2019-01-25:
Same receiver plugged on RC IN on the following boards, all running Release version 3.6.5 Chibios:

PixHawk 1
MindRacer v2
CUAV v5
PixRacer

RC Receiver GR-12L has the capability of outputting either SBUS or SUMD

When set to SBUS all channels work as expected

When set to SUMD RC receiver is not detected by ArduPilot Copter 3.6.5 (tested both Copter and TradHeli) and no channel gets detected.

Checked with Logic analyser the output of the RC receiver and it is outputting data correctly on both modes.

Same receiver with other flight stacks works both on SBUS and SUMD

rmackay9, 2019-01-26:
@lvale, is this a regression from 3.6.4 or something new that was expected to work in 3.6.5 but doesn't?

lvale, 2019-01-26:
didn't work before

rmackay9, 2019-01-26:
Great, thanks.  That's quite a relief.

This is probably something that we should discuss with @tridge

lvale, 2019-03-02:
Adding a Logic trace

with SUMD out
[GR12L_v193_SUMD_OUT_20msPeriod.logicdata.zip](https://github.com/ArduPilot/ardupilot/files/2922031/GR12L_v193_SUMD_OUT_20msPeriod.logicdata.zip)

with SBUS out
[GR12L_v193_SBUS_OUT_20msPeriod.logicdata.zip](https://github.com/ArduPilot/ardupilot/files/2922033/GR12L_v193_SBUS_OUT_20msPeriod.logicdata.zip)

tridge, 2019-03-04:
@lvale I added the data from your test to the RC protocol test code and it passes fine. It get these channel values:
1516, 1500, 1100, 1500, 1900, 1900, 1900, 1900, 1900, 1900, 1900, 1900
so it really should be working!

Ralfde, 2019-05-26:
Would it be easier, if you let SBUS at the LEDSTRIP pin and set SUMD to the RX1 pin?
Then it would be maybe more easy for the software to use the inverted SBUS signal and to SUMD signal at the same time. Also it would be really interesting to switch from SUMD to SBUS an back, in case one of the signals would have failsafe. So RF redundancy could be used.

peterbarker, 2020-11-25:
Please reopen this issue if it is still a problem on our stable or master releases.  tridge was unable to reproduce with the supplied information.

closed 2020-11-25 by peterbarker