ArduPilot/ardupilot #10348: Copter: improve handling when SmartRTL buffer becomes full

rmackay9, 2019-01-26:
SmartRTL has a buffer of up to 500 points but once this is filled up the user simply gets a warning, "SmartRTL: buffer full" and then the mode is no longer usable.  We should improve handling of this situation in three ways:

- improve reporting to the pilot regarding how full the buffer is and possibly enhancements to the ground station to make that more visible.
- a SmartRTL failsafe.  This would trigger when SmartRTL's buffer fills up past a certain percentage of the buffer so users can be more sure that SmartRTL will always be available
- even after the smartRTL buffer fills up we should allows users to enter SmartRTL as long as the vehicle hasn't travelled too far from the point where it filled up.  This would allow users to trigger SmartRTL even after they've been warned the buffer is full.

Naterater, 2019-01-28:
Is it possible to automate the clearance of 50% of the points (every-other point) when the buffer becomes full? I would gladly accept a lower resolution Smart RTL rather than a total loss of points after the buffer filled.

Pedals2Paddles, 2019-01-28:
I was thinking about that concept the other day too.  Problem is that could be the difference between a clear path and a tree.  There would have to be some kind of maximum distance between points that a purge could open up in order to maintain a safe flight path, conditions dependent.

rmackay9, 2019-01-28:
@Naterater, @Pedals2Paddles, yes, we could reduce the points by increasing the value of the [SRTL_ACCURACY parameter](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_SmartRTL/AP_SmartRTL.cpp#L25) and then re-run the simplify step from the beginning (normally it doesn't go back to re-run over points that have already been simplified).  The buffer will still fill up eventually though so eventually the user will have the same issue with the buffer filling up.

I agree though that automatically reducing the accuracy is a good idea too.

mwerezak, 2024-12-01:
In my case I don't expect SRTL to take me all the way to home, just back to where I have a signal again. Would it be possible to add the option to make SRTL use a rolling buffer?