ArduPilot/ardupilot #10350: Rover: 3.5 issues list

rmackay9, 2019-01-26:
This is a list of issues to investigate for the Rover-3.5.0 release:

- OneShot and DShot output not working? Raised by Nando here: https://discuss.ardupilot.org/t/rover-3-5-0-rc2-is-available-for-beta-testing/37731/12

vierfuffzig, 2019-06-19:
@rmackay9 seems the parameter reset bug introduced by https://github.com/ArduPilot/ardupilot/commit/91e5f56 is still present in rover as the corresponding fix https://github.com/ArduPilot/ardupilot/pull/10927/commits/5490ed50ea0182dcaa951e8e59498e6aec81ed0b has only been applied to master.

user report: https://discuss.ardupilot.org/t/mateksys-f405-std-resets-to-default-settings-by-itself/43408

rmackay9, 2019-10-15:
I think I'll close this item because we've started 4.0.0 beta testing and I've moved the one open issue to the 4.0 issues list. https://github.com/ArduPilot/ardupilot/issues/12509

closed 2019-10-15 by rmackay9