ArduPilot/ardupilot #10360: Numerous tests failing autotest for many many builds

jaxxzer, 2019-01-29:
We have currently 10 failing autotests, and tests have been failing for [several months](http://autotest.ardupilot.org/history/).

What is the point of doing these tests if we do not make use of the results? There are obviously problems here that are not being resolved.

I think we need to strive to police this better, increase the visibility of these results, and make some sort of rule about fixing failing tests within some amount of time. Preventing merge of code that fails tests (by adding autotest to ci) would be even better.

Related https://github.com/ArduPilot/ardupilot/issues/9589 and https://github.com/ArduPilot/ardupilot/issues/9592

khancyr, 2019-01-29:
there is something wrong on autotest server as travis is able to pass the test ...
but yes we should add a baner to github that reflect the autotest server status

jaxxzer, 2019-01-29:
travis and autotest are not running the same tests.

khancyr, 2019-01-29:
yes they are ... we only have on set of autotest

jaxxzer, 2019-01-29:
@khancyr FYI

Ci runs build_ci.sh https://github.com/ArduPilot/ardupilot/blob/master/.travis.yml#L37

autotest runs https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/build_autotest.sh which takes several hours to complete.

The problem is not with the autotest server. The problem is our program fails the tests we wrote.

jaxxzer, 2019-01-29:
There are some discrepancies in the tests that are common between travis ci and autotest. For example:
`autotest.py dive.ArduSub`is passing on [travis](https://travis-ci.org/ArduPilot/ardupilot/jobs/486035474#L8849), and failing on [autotest](http://autotest.ardupilot.org/).

jaxxzer, 2019-01-29:
> The problem is not with the autotest server. The problem is our program fails the tests we wrote.

`autotest.py dive.ArduSub` is passing locally also. Maybe I am wrong:

khancyr, 2019-01-29:
Autotest.sh is calling autotest.py like build_ci.sh the test are the same.
But it should'nt take more than 1hour to run... Travis is already quite slow...
I got them running on my company server and that is fine. 
They are some issue on some test, where we are relaying on timeout. Depending on the server, the test or the anwser expected can be too slow or to fast and break the test.
Now, we get the first stone of isolated test. So we will be able to parallelilize them more to finish sooner and see exactly which one are failling.

But I am pretty sure that the problem come from our build server... Something with cache management, like usual... 
And I agree that we should have a banner or feedback on github to allow more people to check if they everything is OK

jaxxzer, 2019-01-30:
> Autotest.sh is calling autotest.py like build_ci.sh the test are the same.

My understanding is that ci is calling `autotest.py` with only a smaller [subset](https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/build_ci.sh#L72) of tests, and autotest is calling `autotest.py` with [all](https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/autotest.py#L704) tests.

peterbarker, 2019-02-05:
I've installed `python-wxgtk3.0` on the server - hopefully that fixes the deps problem.

peterbarker, 2019-02-05:
I've merged a fix for not understanding where MAVProxy's files are being put into master.

peterbarker, 2019-02-05:
On Tue, 5 Feb 2019, Jacob Walser wrote:

> 
> eyyy! @peterbarker it worked! http://autotest.ardupilot.org/
> 🎉
> 
> There remain only two failing tests now.

I've created another PR to fix some remaining issues: 
https://github.com/ArduPilot/ardupilot/pull/10422/commits

peterbarker, 2019-02-06:
All done.  Looking forward to badges so we don't lose track of this again :-)

[image]

closed 2019-02-06 by peterbarker