ArduPilot/ardupilot #10368: stopping runaway flight in copters which have parachutes

robertschulman1, 2019-01-30:
### Feature request

**Is your feature request related to a problem? Please describe.**
When flying a copter, there is worry about a flyaway condition. If the copter has a parachute (mine does), flyaway concerns are addressed with two solutions: (1) use an aux switch to enable parachute release and (2) use the DISARM button in Mission Planner. The problem is that, when operating the aircraft, one needs to hover both over the aux switch (for chute release) and MP's DISARM button.

**Describe the solution you'd like**
By adding a new aux switch function, which I call Aircraft Emergency Stop (or AUXSW_AC_ESTOP), I believe the situation can be addressed. The semantics of invoking this function are to attempt to do both: (1) release the parachute if "conditions are right" (ie. at sufficient altitude, there is a chute and so on) and (2) disarm the motors. If the chute cannot be released, still disarm the motors. By adding this new aux switch function, the operator only need worry about hitting an aux switch in a flyaway situation; hitting MP's DISARM button is no longer needed.

**Describe alternatives you've considered**
Continue using the existing infrastructure was considered. Overloading the semantics of Release Parachute with mandatory motor disarm was also considered.

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
None.
[ac_estop_diff.txt](https://github.com/ArduPilot/ardupilot/files/2814005/ac_estop_diff.txt)

IamPete1, 2019-01-30:
is this not exactly what the existing parachute functionality does? there is already a e-stop function that will stop all motors its function 31.
http://ardupilot.org/copter/docs/parachute.html

robertschulman1, 2019-01-30:
I don’t believe so. My understanding is that opcode 31 only asks for motor cut; what I propose is to perform motor cut for sure, plus, if possible, parachute deployment. In my case, I don’t want chute deployment to occur at too low of an altitude, but above an appropriate threshold, I really would like both the motors to cut and the chute to blow.

 

Does that help explain what I want and why 31 is not good enough?

 

bob

 

http://ardupilot.org/copter/docs/parachute.html

rmackay9, 2019-01-30:
I've changed the subject line a bit to clarify the change a bit.  I think it's reasonable especially if we can think of a good name name for the switch option.  Maybe "Parachute OR Disarm" would be good.

IamPete1, 2019-01-30:
Does function 22 - Parachute Release not do this? I guess it must ignore everything else and always deploy the chute, rather than checking the altitude?

I wonder if we could make RC_Options do more than one thing action, make them sort of a bit mask. So you could stack multiple things on the one switch. Probably wouldn't actual be useful for many of the options tho.

peterbarker, 2019-01-31:
On Wed, 30 Jan 2019, bob schulman wrote:

> By adding a new aux switch function, which I call Aircraft Emergency Stop (or AUXSW_AC_ESTOP), I believe the situation can be addressed. The semantics of

I'm thinking an AUXSW_ASSUME_CRASHED.

We'd do everything we'd do if the aircraft was out-of-control (see 
Copter's crash_check.cpp) - that stops the motors and deploys the 
parachute, and should sound the lost-Copter alarm and deploy the 
fire-retardent foam.

robertschulman1, 2019-01-31:
I’m thinking that if the aircraft is crashed, there wouldn’t be much of a need of chute deployment ;-> As you point out, there’s already code which handles that case.

 

My chief motivation for this change is either a runaway aircraft or one which is nearly, or truly out of control.

 

bob

robertschulman1, 2019-01-31:
Function 22 does do this *if* all the gateway checks pass (eg. if the aircraft is at sufficient altitude). Function 22 does nothing if any of the gateway checks fail. What I am asking for is: even if the chute doesn’t release, at least stop the motors.

 

bob

robertschulman1, 2019-01-31:
Hmmm. I guess I am missing something as I’m new to all of this. What is the command one can send from the ground to the aircraft to tell the aircraft, “consider yourself crashed and deploy the parachute and stop the motors”?

 

bob

 

Yep - my point here is that we have code that handles that case - we 
should consider it "crashed" - which is supposed to do as much as possible 
to make the situation safe.