ArduPilot/ardupilot #10369: GPS-PPS tracking

Naterater, 2019-01-30:
### Feature request

**Is your feature request related to a problem? Please describe.**
High timing accuracy is needed for certain applications. GPS units use highly accurate timing for position measurement, and PPS (an output signal at a precise time) is supported on high end units for time synchronization. The PPS event can currently be logged as a camera feedback event, however that removes the true camera feedback ability.

**Describe the solution you'd like**
1. Allow an additional feedback pin to be logged in addition to the cam feedback
2. Use this PPS signal to synchronize the system clock and avoid drift

**Describe alternatives you've considered**
An external IMU/Logging system for photogrammetry

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[X] Rover
[  ] Submarine

**Additional context**
https://github.com/ArduPilot/ardupilot/issues/7892

adpjhu, 2024-02-28:
It appears as if this has been implemented [here](https://github.com/ArduPilot/ardupilot/commit/9834304525caf7bb36eb7c8bab62506571abe5a7) . Should this issue be closed?

magicrub, 2026-01-02:
Yeah, fairly odd that it's a ublox-only-ish type thing. That pin ISR should be a generic thing for all drivers. But, it's only logging it. There's still no action taken for any PPS event that I know of. I can see how that might be useful in the code but it would be hard to do with zero-ish latency. To make it really useful we would need a user-definable function inside the gps PPS ISR itself which comes with all kinds of crazy risk

vidma, 2026-02-15:
1. alright so I **implemented some quick/simple PPS support for NMEA, as well as logging to blackbox/binary logs**. Maybe it's not super accurate as it still uses a regular interrupt  on a rising PPS signal on GPIO which fetches current sys time (like on UBLOX), but gives consistent results, so I'd suppose it's good enough.

on https://micoair.com/flightcontroller_micoair743lite/ , according to PPS from LG580P, I was consistently getting around `~0.17%` faster clock than it should. That seems a quick a huge clock mismatch to me -  I guess due to a cheap low-quality 8MHz crystal with a large static offset used on this cheap board.

2. **Do you think it would make sense to correct the times returned by `hrt.c`  like `hrt_micros64()`, or it could do more harm than good?**  For now, I'm only considering static/hardcoded into hwdef correction-factor (instead of dynamic/real-time based on PPS), to make sure the timestamps stay monotonously increasing, as otherwise weird things may happen and it'd need more complex logic to make sure monotonicity and thread-safety.

Now by applying a static factor in hwdef, the timedrift seem to have dropped down to < `0.002%`  (only when compared with hrt.c , of course - the  timing peripherals and physical timers is still off as before):

[image]

for example:
hwdef:

```python
#  undef and define PPS pin
undef PD14
PD14  GPS_PPS INPUT FLOATING GPIO(64) #M15 - any free GPIO works fine here
define HAL_GPIO_PPS 64

define AP_CLOCK_CORRECTION_PPM 1700
```

AP_GPS_NMEA.cpp:

```cpp
#ifdef HAL_GPIO_PPS
void
AP_GPS_NMEA::pps_interrupt(uint8_t pin, bool high, uint32_t timestamp_us)
{
    uint64_t now_us = AP_HAL::micros64();
    uint64_t earlier_pps_time_us = _last_pps_time_us;
    _last_pps_time_us = now_us;

    // FIXME: this is possibly rather nasty to have logging inside interrupt, but let's keep things simple for now
    AP_GPS::get_singleton()->Write_PPS(now_us, earlier_pps_time_us);
}
#endif // HAL_GPIO_PPS
```

hrt.c:

```diff
diff --git a/libraries/AP_HAL_ChibiOS/hwdef/common/hrt.c b/libraries/AP_HAL_ChibiOS/hwdef/common/hrt.c
index f1e510cceb..9fdd3af0a6 100644
--- a/libraries/AP_HAL_ChibiOS/hwdef/common/hrt.c
+++ b/libraries/AP_HAL_ChibiOS/hwdef/common/hrt.c
@@ -27,6 +27,33 @@

 #include "../../../AP_Math/div1000.h"

+#ifndef AP_CLOCK_CORRECTION_PPM
+#define AP_CLOCK_CORRECTION_PPM 0
+#endif
+
+static inline uint64_t clock_correct_us(uint64_t us)
+{
+#if AP_CLOCK_CORRECTION_PPM > 0
+    // clock is fast: return a slightly smaller time
+    return us - (us * (uint64_t)AP_CLOCK_CORRECTION_PPM) / 1000000ULL;
+#elif AP_CLOCK_CORRECTION_PPM < 0
+    // clock is slow: return a slightly larger time
+    return us + (us * (uint64_t)(-AP_CLOCK_CORRECTION_PPM)) / 1000000ULL;
+#else
+    return us;
+#endif
+}
+
+static inline uint32_t clock_correct_us32(uint32_t us)
+{
+#if AP_CLOCK_CORRECTION_PPM != 0
+    return (uint32_t)clock_correct_us((uint64_t)us);
+#else
+    return us;
+#endif
+}
+
+
 /*
   we have 4 possible configurations of boards, made up of boards that
   have the following properties:
@@ -64,7 +91,7 @@ static uint32_t system_time_u32_us(void)
 #if CH_CFG_ST_FREQUENCY != 1000000U
     now *= 1000000U/CH_CFG_ST_FREQUENCY;
 #endif
-    return now;
+    return clock_correct_us32(now);
 }
 #else
 #error "unsupported timer resolution"
@@ -93,7 +120,7 @@ static uint64_t hrt_micros64I(void)
 #ifdef AP_BOARD_START_TIME
     ret += AP_BOARD_START_TIME;
 #endif
-    return ret;
+    return clock_correct_us(ret);
 }

 static inline bool is_locked(void) {
```