ArduPilot/ardupilot #10376: Fail to build AP_Motors_test example with CubeBlack

thethaibinh, 2019-01-31:
### Bug report

**Issue details**
When I tried to buil AP_Motors_test example from AP_Motors library with configure board=CubeBlack, I got the error: 
"No param area found 
Error: Param defaults support not found in firmware"

**Version**
AP Motors Library

**Platform**
[  ] All

**Hardware type**
CubeBlack

BrightSoulXYHY, 2019-03-06:
I have the same problem when build AP_Notify_test with configure board=Pixhawk1&fmuv2 and so on. How about build other Library?

IamPete1, 2021-02-08:
Sorry we went able to help, Both `./waf --target examples/AP_Motors_test` and `./waf --target examples/AP_Notify_test` are working for me on CubeBlack now so I'm going to close this.

closed 2021-02-08 by IamPete1