ArduPilot/ardupilot #10389: StorageAccess initialization

azdludovic, 2019-02-01:
### Bug report

**Issue details**

In [StorageManager.cpp](https://github.com/ArduPilot/ardupilot/blob/master/libraries/StorageManager/StorageManager.cpp), the address of _StorageManager::layout_ is statically initialized to _layout_default_., and change to _layout_copter_ only in the _Copter::setup()_ method.

However, at least for _AC_PolyFence_loader_ and _AP_Param_, the associated _StorageAccess_ is also statically initialized. As the _StorageAccess_ uses _layout_ to determine the size of the available storage, the value returned by _storage.size()_ is the one computed with _layout_default_. The _size()_ method of StorageFence will for example return 672 instead of 560.

Therefore, in _AC_PolyFence_loader_, some points won't be saved or load even though the load/save will return true (the only check uses _max_points()_, which uses _storage.size()_). This might causes arithmetic exception in case the Fence is reloaded afterwards as the _load_point_from_eeprom_ method won't initialize the _point_ value and an exception might occur in _location_diff()_ [AC_Fence.cpp@565](https://github.com/ArduPilot/ardupilot/blob/2834fd8017a03b661983d8d0cf107f4d2df0df55/libraries/AC_Fence/AC_Fence.cpp#L565)

**Version**
ArduCopter 3.5

**Platform**
[ ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
Found on a Cube, applies to all hardware

khancyr, 2019-02-01:
Beside the fact that the instantiation is risky since we are dealing with static storage, the error is valid.
when changing the layout we need to update the storage.size()

IamPete1, 2021-09-12:
Closed by https://github.com/ArduPilot/ardupilot/pull/13299 we now have consistent init ordering.

closed 2021-09-12 by IamPete1