ArduPilot/ardupilot #10428: Rangefinder filtered distances SONR.Dist is always jumping from correct to zero values

acobo, 2019-02-06:
### Bug report

**Issue details**

We have tested a Terabee Teraranger EVO 60M rangefinder connected to a Pixhawk v1 autopilot (using the I2C bus). Looking at the dataflash logs, the values of RFND.Dist1 seem to be OK, but the values of SONR.Dist  are jumping between a correct distance and a zero value. That is, for every correct sample, there is one (sometimes more consecutive) zero values. According to @magicrub :

> the value in RFND.Dist1 is the raw value form the rangefinder, it is not tilt compensated. For Plane, the compensated and filtered value that is actually used can be found in SONR.Dist"

(issue https://github.com/ArduPilot/ardupilot/issues/2377)

So it seems that the raw distances from the rangefinder are Ok, while the filtered and corrected values are corrupted.  Any help with this issue will be much appreciated. We need the rangefinder to assist with the automatic landing, but we are not sure at the moment if the autopilot is using it or not. According to the source code, I understand that we should get this mavlink message in the ground station:

`gcs().send_text(MAV_SEVERITY_INFO, "Rangefinder engaged at %.2fm", (double)rangefinder_state.height_estimate);`

But we are not seeing it. Thankyou in advance.
[image]

[image]

**Version**
Arduplane  3.9.4 & 3.9.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
BORMATEC Maja

**Hardware type**
Pixhawk v1 with a teraranger EVO 60M rangefinder connected using I2C

**Logs**
https://www.dropbox.com/s/s55wv1mp9im1iue/dataflash%20log%20teraranger.BIN?dl=0

acobo, 2019-02-07:
Hello again,
 we have verified that another problem we had, that the MAVLINK_VFR_HUD message with ground and airspeed was stopped to be sent to the ground station a few seconds after takeoff (more details in this post:  https://discuss.ardupilot.org/t/mavlink-msgs-with-hud-information-are-not-sent-for-a-few-minutes-after-takeoff/37958) has been solved once we removed the teraranger rangefinder from the I2C bus.
We believe that there is an issue with the driver of this rangefinder and the I2C bus. Some other people have reported a kind of interference with other devices in the I2C bus, maybe with a previous versión of the driver, I don't know.  Maybe the fact that it takes 500us to reply as a slave is too much time and it is affecting the polling of other devices in the bus?

Thanks in advance, best regards.

acobo, 2019-03-08:
Hello again,
 I have just discovered that someone has found the cause of this problem:

https://discuss.ardupilot.org/t/teraranger-one-cause-bad-lidar-health/23124/3

The function update() in ArduPlane.cpp is called at 50Hz:

`SCHED_TASK(read_rangefinder,       50,    100),`

While the timer() function in the driver  AP_RangeFinder_TeraRangerI2C.cpp to actually do the measurement is called at 20Hz:

`dev->register_periodic_callback(**50000**,
                                    FUNCTOR_BIND_MEMBER(&AP_RangeFinder_TeraRangerI2C::timer, void));`

 The problem is when there is no a new measurement available, which will occur 3/5th of times, the code in the update() function changes the state to "NoData":

`void AP_RangeFinder_TeraRangerI2C::update(void)
{
    WITH_SEMAPHORE(_sem);
    if (accum.count > 0) {
        state.distance_cm = accum.sum / accum.count;
        state.last_reading_ms = AP_HAL::millis();
        accum.sum = 0;
        accum.count = 0;
        update_status();
    } else {
        set_status(RangeFinder::RangeFinder_NoData);
    }
}`

The driver implementation of other rangefinders only changes the status to NoData if more than 200ms have passed without a new measurement, otherwise, it just does not update the distance, for example for LeddarOne:

`
void AP_RangeFinder_LeddarOne::update(void)
{
    if (get_reading(state.distance_cm)) {
        // update range_valid state based on distance measured
        state.last_reading_ms = AP_HAL::millis();
        update_status();
    } else if (AP_HAL::millis() - state.last_reading_ms > 200) {
        set_status(RangeFinder::RangeFinder_NoData);
    }

}`

I have manually count from the log screen above and, for every 50 measurements, exactly 30 are zero, which I think it is a proof that this is the issue. This issue does not happend with Copter because the update task has the same frequency, 20Hz. 

I think that the solution could be to increase the frequency of the callback to ::timer() to 50Hz or more (maximum for the Teraranger EVO 60 we have is 240Hz), and, just in case, doing the call to set_status(RangeFinder::RangeFinder_NoData) only after 200ms, like with the others rangefinder.

I would be very grateful if developers could create a commit to resolve the issue in next versions.
Thanks!

rmackay9, 2019-03-08:
@acobo,

Very nice investigation.  So, yes, I think we should change the driver so that it uses the same 200ms check as other drivers.

WickedShell, 2019-04-09:
Fixed in #10780

closed 2019-04-09 by WickedShell