ArduPilot/ardupilot #10435: Division-by-zero error for two filter parameters

ktskt, 2019-02-06:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**
If I put some values larger than or equal to 0 and 'lower' than 1 on  INS_GYRO_FILTER and INS_ACCEL_FILTER, the program will crash due to division by zero.

**Version**
3.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[O] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
But, other must share the same problems

**Airframe type**
Quadcopter but applicable to other frames

**Hardware type**
SITL

peterbarker, 2019-02-07:
I'm thinking we should have some boot-time checks for out-of-bounds parameters like this and enter `sensor_config_error` loop (perhaps with a better name).

peterbarker, 2019-02-07:
... in the meantime, "well don't do that then!" :-)

ktskt, 2019-02-07:
@peterbarker Thank you for your comment.
Of course, I won't do that. It is a just bug report that I found :).
In the meantime, I think it would be better to change related document ranges since both parameters can be 0 according to the document. 

http://ardupilot.org/copter/docs/parameters.html#ins-gyro-filter-gyro-filter-cutoff-frequency
http://ardupilot.org/copter/docs/parameters.html#ins-accel-filter-accel-filter-cutoff-frequency

peterbarker, 2019-02-07:
On Thu, 7 Feb 2019, ktskt wrote:

> In the meantime, I think it would be better to change related document
> ranges since both parameters can be 0 according to the document.

Ah.  Right.  Documented behaviour.  I guess we need to fix the behaviour 
rather than paper over people's parameter settings...

Similar code sets alpha to 1.0 - but compute_params is a little 
more complicated than that....

Reproduction is simple; start SITL and  param set INS_GYRO_FILTER 0

@tridge

PR #10456 (2019-02-08): Filter: fixed crash on zero cutoff frequency
this fixes #10435 
ping @peterbarker
commit 6c09b268 (2019-02-08): Filter: fixed crash on zero cutoff frequency

this fixes #10435
commit 0b890a7b (2019-02-09): Filter: fixed crash on zero cutoff frequency

this fixes #10435
commit 30d11f6a (2019-02-09): Filter: fixed crash on zero cutoff frequency

this fixes #10435
closed 2019-02-08 by tridge