ArduPilot/ardupilot #10436: Add user text field to F7 OSD for aircraft name or ham callsign.

atshort, 2019-02-06:
Could we have a short text field on the F7 OSD for the user to specify a vehicle name or callsign?

This would be useful for identifying FPV video recorded from multiple vehicles.

Also possibly needed for a ham callsign in some locations.

**Describe the solution you'd like**
How about an 8 character text field added to the OSD parameters.

**Describe alternatives you've considered**
None.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[x] Rover
[  ] Submarine

Thank you.

Rusty105, 2019-04-09:
I will second this. The stand alone OSDs have this, and as stated above, is needed for compliance with amateur radio regulations in some areas.

Rusty105, 2019-04-25:
I have given this a shot at implementing, I'll post updates when I get something.

squilter, 2021-02-11:
Callsign support was added a few months ago with #15789. I think that this issue can be closed.

atshort, 2021-02-11:
Done. Thank you.

closed 2021-02-11 by atshort