ArduPilot/ardupilot #10437: Feature Request: Parameter to display wp_dist regardless of flight mode

Chase-Tillman, 2019-02-06:
**Issue Details**

I am attempting to optimize my inspections of energized towers through the use of a displayed distance from the tower. 

**Feature Request**

 Add a parameter that could force "wp_dist" to be shown regardless of flight mode.

**Considered Alternatives**

The three distances configured to be displayed through telemetry are: wp_dist, DistToHome, and GeoFenceDist. 

- GeoFenceDist would not behave correctly, likely due to the copter not being within the inclusive geofence. 

- DistToHome will not work without some way to lock the home position to a set location. With the ability to lock the home location, rally points could be used and the problem would be worked around.  Re-loading a mission with a set home location while in mid-flight displays the correct distance through the GCS, but not through the OSD. This method is also unusable for safety reasons.

- WP_Dist is the best workaround I have. I am able to toggle auto flight mode to view the distance, but this is not a viable strategy for consistently maintaining the proper standoff distance.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2019-02-07:
Which mode are you working in?

Which particular mavlink message are you refering to?  Assuming it is `NAV_CONTROLLER_OUTPUT` - we fill that in with all sorts of different things depending on the mode you're in.  In Circle it's the centre of the circle.  In follow mode it's the distance to the vehicle you are following.  In auto... well, I think you can imagine ;-)

There's no particular reason to send auto's waypoint distance when you're in stabilize mode.  It would be just as (in)valid to send through the distance to the system you're set to follow when in follow mode (and might be very handy, actually...).

peterbarker, 2019-02-07:
... one option might be to populate the wp_dest field from the last mode you were in that supported updating that field.  That could be quite confusing, however.

Chase-Tillman, 2019-02-07:
> Which mode are you working in?
>
> Which particular mavlink message are you refering to? Assuming it is `NAV_CONTROLLER_OUTPUT` - we fill that in with all sorts of different things depending on the [...]

"Which mode are you working in?"
Loiter mode

"Which mavlink message?"
I am still extremely ignorant to ardupilot (and coding in general), but am trying to learn as much as possible. I believe you are correct that it is NAV_CONTROLLER_OUTPUT, given wp_dist is found within, but if a definitive answer for this is necessary, I need to learn more before requesting this.

The solution to my problem may not be found within wp_dist, but instead the creation of a new parameter that inputs GPS coords (or a WP) and calculates the distance from the copter to the coords the same as wp_dist does during auto, but not limited to a flight mode. Unfortunately I would not know where to begin in requesting this. 

Feel free to close this request given a lack of details to proceed.

peterbarker, 2019-02-07:
On Wed, 6 Feb 2019, Chase-Tillman wrote:

> The solution to my problem may not be found within wp_dist, but instead the
> creation of a new parameter that inputs GPS coords (or a WP) and calculates
> the distance from the copter to the coords the [...]

Interesting thought.

If the vehicle isn't actually using this position for navigation - could 
it be done off-board instead?  Naturally if this is something like an OSD 
you've got limitations there :-)

Chase-Tillman, 2019-02-07:
> Interesting thought. If the vehicle isn't actually using this position for navigation - could it be done off-board instead? Naturally if this is something like an OSD you've got limitations there :-)

A co-worker is attempting to calculate this using an arduino and input the value as possibly the battery current or voltage, or some other field that is currently displayed by the OSD - replacing the previous value. We will see if this workaround works out.