ArduPilot/ardupilot #10448: Copter/Heli: Matek F405 Wing I2C LED issue 3.6.5

Fi156, 2019-02-07:
Hey guys,

**related**
https://discuss.ardupilot.org/t/copter-3-6-5-released/37733/11
https://github.com/ArduPilot/ardupilot/issues/9498

**Issue**
With Heli 3.6.3 compass and LED worked out of the box!
(Never expected that, because of missing LED connection, see below.)
After updating to Heli 3.6.5 system halts completely after boot.
(After short blinking, red led is lid constantly, no connection over UART telemetry possible)

NTF_LED_TYPES:
3.6.3: 199, everything worked
3.6.5: If set to 0, AP boots normally, compass works, LEDs on FC are all constantly lid, LED on GPS module is off

Following setup:
Connected: UART for GNSS, I2C for Compass, 5v, 3.3V, GND
Not connected: LED, Buzz, Key

See pinout GNSS:
http://www.holybro.com/manual/Pixhawk4-GPS-Quick-Start-Guide.pdf

**Version**
3.6.5 HELI

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
HELI

**Hardware type**
Matek F405 Wing 
Holybro Pixhawk 4 GPS, M8N Module with integrated LED, Compass and Button

**Logs**
no logs

rmackay9, 2019-02-08:
This sounds similar to the issue reported by [Maciek here in the Copter-3.6.5 release thread](https://discuss.ardupilot.org/t/copter-3-6-5-released/37733/10).

rmackay9, 2019-02-26:
I've reproduced a (perhaps related) freeze up issue if the HMC compass is connected.

IamPete1, 2021-08-15:
This sounds like the i2c interrupt storm that has been resolved in https://github.com/ArduPilot/ardupilot/pull/12134, please re-open if the issue remains on 4.1

closed 2021-08-15 by IamPete1