ArduPilot/ardupilot #10450: SITL: missing library in sim_vehicle.py (Windows)

mikerob, 2019-02-07:
**Issue details**

Running the latest version of SITL from a complete pull on 07 Feb 19 on Windows results in the following error:

```
Traceback (most recent call last):
  File "../Tools/autotest/sim_vehicle.py", line 25, in <module>
    from MAVProxy.modules.lib import mp_util
ImportError: No module named MAVProxy.modules.lib
```

**Version**
07 Feb 19 full download - happens on all vehicle types 

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

The problematic line is:

from MAVProxy.modules.lib import mp_util

Removing this line lets it run fine.

It was added as part of this PR:

https://github.com/ArduPilot/ardupilot/commit/ec8cf10f7c44829294db70d9aefe40154012fd7c

IamPete1, 2019-02-08:
do you have MAVProxy installed?

nagendra092918, 2019-02-12:
same problem i installed MAVProxy but getting same error how to slove that please help
Traceback (most recent call last):
  File "/home/nagendra/ardupilot/Tools/autotest/sim_vehicle.py", line 25, in <module>
    from MAVProxy.modules.lib import mp_util
ImportError: No module named MAVProxy.modules.lib

magicrub, 2019-02-12:
It might just be that mavproxy needs another release.
@peterbarker

peterbarker, 2019-02-13:
@stephendade looks after our Windows releases for MAVProxy.  We do run them through appveyor, so...

The error really looks like ArduPilot's Python code can't see MAVProxy.  @nagendra092918 @mikerob which instructions did you follow to get both MAVProxy and ArduPilot installed for Windows?

@nagendra092918 ARE you running on Windows?!

mikerob, 2019-02-13:
@peterbarker, I think it was the "standard" instructions: https://ardupilot.github.io/MAVProxy/html/getting_started/download_and_installation.html

It was working for a previous code base (from October last year), and it works if I remove that line.  

@magicrub My Mavproxy dates back to October as well - is an update required?

stephendade, 2019-02-13:
I've replicated that behaviour on my Windows PC - it's indeed an issue with Ardupilot.

@peterbarker - the issue is that the MAVProxy installer uses a compiled mavproxy.exe, not the *.py files. Thus anyone using the MAVProxy installer will encounter this issue.

I'd be against assuming that MAVProxy is installed when running ``sim_vehicle.py``, as some users do use alternative GCS's...

stephendade, 2019-02-13:
Also, Appveyor simply builds MAVProxy. There's no tests or checks run.

peterbarker, 2019-02-13:
On Tue, 12 Feb 2019, Stephen wrote:

> I've replicated that behaviour on my Windows PC - it's indeed an issue
> with Ardupilot.
> 
> @peterbarker - the issue is that the MAVProxy installer uses a compiled
> mavproxy.exe, not the *.py files. Thus anyone using the MAVProxy installer
> will encounter this issue.

So if we change things up to look for a mavproxy.exe if we can't find a 
mavproxy.py it might work?

Could you try hacking up sim_vehicle.py to just look for the .exe and see 
if it all hangs together, please?

What alternatives do people have to get going if we don't hack it up to 
look for a .exe?

> I'd be against assuming that MAVProxy is installed when running
> sim_vehicle.py, as some users do use alternative GCS's...

Weeeelllll.. VERY few people, given "headless" has only been in for a 
few of months.

stephendade, 2019-02-13:
> So if we change things up to look for a mavproxy.exe if we can't find a mavproxy.py it might work?

Nope. ``sim_vehicle.py`` requires a MAVProxy library (``MAVProxy.modules.lib.mp_util``). The process by which pyinstaller turns MAVProxy into an exe means that any libraries contained within it are inaccessible to any external programs. I think it only has the byte-compiled *.pyc files

>Weeeelllll.. VERY few people, given "headless" has only been in for a few of months.

The ``--no-mavproxy`` argument has been in there for some time :)

peterbarker, 2019-02-13:
On Wed, 13 Feb 2019, Stephen wrote:

> So if we change things up to look for a mavproxy.exe if we
>       can't find a mavproxy.py it might work?
>
> Nope. sim_vehicle.py requires a MAVProxy library
> (MAVProxy.modules.lib.mp_util). The process [...]

Well, that's pretty damning, then.

>       Weeeelllll.. VERY few people, given "headless" has only been
>       in for a few of months.
> 
> The --no-mavproxy argument has been in there for some time :)

Huh.  About 2 years.  That's still measurable in months :-P

  Committed two years ago - 'though I think merged much more recently 
than that.

Turns out the relevant functions are also in mavextra.py:

https://github.com/ArduPilot/ardupilot/pull/10511

peterbarker, 2020-11-25:
Pretty sure our Windows installs are working again now-adays.  Please reopen if it is an issue.

closed 2020-11-25 by peterbarker