ArduPilot/ardupilot #10465: Allow servos to have mixed output types

magicrub, 2019-02-09:
MOT_PWM_TYPE controls PWM signal type to different motors. This includes DShot, "RC PWM", and brushless PWM (duty cycle). We don't have a way to map, for example, brushless PWM signal to an aileron or ICE starter or Gripper. We need to de-couple PWM type to servo function. Problems we need to watch out for is you can damage a traditional servo if you send brushed/duty PWM when it expects normal/RC PWM pulses. This is especially bad because even if we configured it correctly, the configuration is hardware dependant so one configuration may work on one autopilot HW, but will kill servos if same settings are used on another HW.

For example, here's the [map for a Cube](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/CubeBlack/README.md#pwm-output). There's 3 PWM groups which are tied to timer mappings. Right now there is no source code to be timer-awar, that is abstracted out in the HAL for a just-works solution but with limited visibilty.

- [ ] Rethink and/or remove MOT_PWM_TYPE so that a Normal/Brushed applies to any SERVOx_FUNCTION, not limited to just motors.
- [ ] Add new param (temporary name) SERVO_TYPE that is a bitmask that enables PWM duty (brushed) on any servo output.
- [ ] Add an always-on arming check (or completely mute servo outputs) on SERVO_TYPE configuration to ensure the configuration matches the HW's capabilities so that the configured behavior will be actual behavior.

IamPete1, 2019-02-09:
Also consider that for rovers you might want to use brushed with relay function to allow reversing, then you need to associate two pins to each function.

magicrub, 2019-02-10:
This will be a better solution for https://github.com/ArduPilot/ardupilot/issues/8632