ArduPilot/ardupilot #10466: Copter: speed up EKF failsafe by checking if velocity innovations > 2x FS_EKF_THRESH

rmackay9, 2019-02-09:
Copter's EKF failsafe checks if 2 out of 3 innovations (compass, velocity, position) are greater than the FS_EKF_THRESH parameter value.  If they are (for more than a second or two) the failsafe triggers.

In cases where the GPS position/velocity has gone bad, we can likely speed up triggering of the EKF failsafe if we also trigger in cases where the velocity innovations are greater than twice the FS_EKF_THRESH.

Here is a graph from one log in which the compass was fine, but the GPS was providing terrible positions and velocities.  In this case the GPS itself also reported that it had very little idea of where it was.  Despite this it took 13seconds for the EKF failsafe to trigger.

[image]

chobitsfan, 2019-03-12:
I may have a similar situation. Our copter was jammed in a power tether test and it went crazy. After inspecting log, I found SV quickly go above 2 but SP and SM did not go above 0.6 Therefore it did not trigger EKF failsafe.
[image]

rmackay9, 2019-03-12:
@chobitsfan, thanks for the report.  It's not a super hard fix so hopefully one of us can find the time to work on it.. certainly we should get this included in Copter-3.7

chobitsfan, 2019-03-12:
hi @rmackay9 maybe I could create a PR for this issue? Currently, ekf_check_state.fail_count increased when  2 out of SV, SP and SM. Maybe ekf_check_state.fail_count also increased when either SV, SP or SM is greater  then 2x FS_EKF_THRESH ?

rmackay9, 2019-03-12:
@chobitsfan, I was hoping you'd offer, yes please!

PR #10801 (2019-03-15): Copter: speed up EKF failsafe by checking if velocity innovations > 2x FS_EKF_THRESH
fix #10466 

I tested it with skyviper journey GPS and attack it using GPS jammer. After GPS is jammed, my drone went crazy but when SV go up to twice FS_EKF_THRESH, EKF failsafe is triggered and landing. In my test, SP and SM did not go over 0.8 (default  FS_EKF_THRESH), so EKF failsafe was not triggered in current master.
[image]
[image]
closed 2019-03-15 by rmackay9