ArduPilot/ardupilot #10467: Plane: SITL RCx_OPTION invalid number results in no boot

IamPete1, 2019-02-09:
### Bug report

**Issue details**

Plane SITL using real flight, on master change any RC option to 31, (may also work with other invalid numbers). SITL continues to work fine, gives a invalid option warning. But reboot and SITL will seem to be running but cannot be connected to with MP or MavProxy. Only way to fix is to do a distclean and parameter clear.  Not tested IRL.

Encountered during testing of https://github.com/ArduPilot/ardupilot/pull/10290 

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
any

**Hardware type**
SITL

**Logs**
-

peterbarker, 2019-02-09:
On Sat, 9 Feb 2019, Peter Hall wrote:

> Plane SITL using real flight, on master change any RC option to 31, (may also work with other invalid numbers). SITL continues to work fine, gives a
> invalid option warning. But reboot and SITL will [...]

This won't happen IRL.

This is a check designed to catch someone creating a new function and not 
handling the initialisation for that function.

AP_HAL::panic("RC function (%u) initialisation not handled", ch_option);

That string should have been visible somewhere to you....

Starting sim_vehicle with the "-w" option will wipe your parameters.  That 
should be sufficient to continue.

IamPete1, 2019-02-09:
Ah, that would explain it, I guess I should get the panic message in the second SITL window thing that shows the frame rate that I don't get / haven't worked out how to set up using cygwin.

IamPete1, 2019-02-10:
can confirm properly initializing the function mends this, Does anyone know if its possible to get the extra SITL information window using Cygwin?

closed 2019-02-10 by IamPete1

peterbarker, 2019-02-10:
On Sat, 9 Feb 2019, Peter Hall wrote:

> can confirm properly initializing the function mends this, Does anyone know if its possible to get the extra SITL information window using Cygwin?

I can't do that, but you should still be able to get that output from a 
file in /tmp - it should tell you the filename...

https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/run_in_terminal_window.sh#L47

IamPete1, 2019-02-10:
Thank You, I know where to look if I get this odd behavior again