ArduPilot/ardupilot #10473: Add default TRIM_ARSPD_CM (default) parameter or active airspeed set point

custeve, 2019-02-10:
### Feature request

**Is your feature request related to a problem? Please describe.**
This applies to ArduPlane, in operations where the airspeed is varied, the trim airspeed set point does not reset for the following flight. The concept of a Trim Airspeed is ideal for mission profiles that use a consistent airspeed for all operations, but poor for when operations require numerous GCS initiated or programmatic (ie DroneKit or other) airspeed set point changes. 

**Describe the solution you'd like**
There are two possible solutions, the preferred is that TRIM_ARSPD_CM would be the DEFAULT airspeed determined during flight tuning (thus, it's appropriately in on the BASIC tuning page of Mission Planner). In flight, the current target airspeed would initialize to TRIM_ARSPD_CM, but changes to the target airspeed would result from the MSG_CMD_DO_CHANGE_SPEED command, not a parameter change. It is debatable, for example, that a change to RTL would revert to the default airspeed. 

Ideally, this current airspeed set point would be output in a MAVLINK message for monitoring by a GCS or other program (ie DroneKit). 

**Describe alternatives you've considered**
An alternate solution would be to add a parameter for the default trim airspeed, then have TRIM_ARSPD_CM operate as it currently does. This may maintain backward compatibility with Mission Planner, however the operation reliance on parameters is somewhat clunky. 

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Additional context**
None.

magicrub, 2019-02-10:
yes, I agree. The DO_CHANGE_SPEED command should not override the param, it should override the param until next boot, or perhaps until you leave AUTO/GUIDED modes.

WickedShell, 2019-02-11:
I'm not following this well. We do support DO_CHANGE_SPEED, yes it writes a value to the parameter, but the value doesn't persist across a reboot, and until we've gotten a new DO_CHANGE_SPEED this is the correct option.... And since there is no command for revert to default speed I don't see the problem. Could someone explain it better?

I agree there should be better emission of target airspeeds. It can be derived by use of NAV_CONTROLLER_OUTPUT + VFR_HUD but there are timing/race problems with doing that, it doesn't give you the precise setpoint.

IamPete1, 2021-08-08:
closed by https://github.com/ArduPilot/ardupilot/pull/16719

closed 2021-08-08 by IamPete1