ArduPilot/ardupilot #10487: Add climb/descend rate to LOITER_TO_ALT

custeve, 2019-02-11:
### Feature request

**Is your feature request related to a problem? Please describe.**
When using an aircraft for atmospheric profiling, it is desired to control the climb/descent rate for the mission to be much slower than the TECS set rates. Changing the rates globally would produce undesired results otherwise from the mission. This feature could also prove beneficial for controlling helical landing approaches. 

**Describe the solution you'd like**
Add a parameter to the LOITER_TO_ALT mission command to set the desired climb and descend rate. This rate would only apply to that active mission item and would not override, nor be able to exceed, the TECS asc/dsc rates. 

**Describe alternatives you've considered**
An alternate to the mission command (or a way to compliment and augment the mission command) would be to add an entirely new mode, something like LOITER_MOD_ALT. This mode would require parameters to set the ceiling and floor, and the rates. A sub-mode could instruct the aircraft to repeat, stay at ceiling or stay at floor. The advantage to this implementation is that profiling could be performed without a ground station, just a mode change on the RC handset.

After describing this, the alternate solution seems ideal for our application, specifically. 

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Additional context**
It appears that Copter implements this to some degree. So, it may be just applicable to Plane.

magicrub, 2019-02-11:
This make sense, not sure why anyone thought of it before! I've needed this and had to do silly stuff to get the same effect. This would make Helical landing much easier to plan since the descent rate will be in the mission and not tied to params.

Thanks @custeve !!! I feel like doing this now now now

custeve, 2019-02-11:
@magicrub - I would not complain one bit about having this functionality sooner rather than later! We have a deployment late spring (we storm chase with UAS), and if I could test this functionality by April, it would save having to add the functionality through a DroneKit ground station (put that in the silly stuff column I was already planning on, but would love to not need to do).

rmackay9, 2019-02-12:
By the way, Copter's DO_CHANGE_SPEED supports setting the target climb and descent rate..

magicrub, 2019-02-12:
@custeve here's a PR for this. https://github.com/ArduPilot/ardupilot/pull/10493 Do you have the ability to test this on SITL and on real aircraft?

magicrub, 2019-02-12:
This PR is plane only.