ArduPilot/ardupilot #10502: AP_RangeFinderParams: RNGFND not collapsed if NONE

olliw42, 2019-02-12:
@peterbarker 
in this line
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_Params.cpp#L13
you probably want(ed) to use
`AP_GROUPINFO_FLAGS("TYPE",    1, AP_RangeFinder_Params, type, 0, AP_PARAM_FLAG_ENABLE),`
instead

(the 0 should be replaced with RangeFinder_TYPE_NONE, this would in addition requires moving the RangeFinder_Type enum to AP_RangeFinderParams.h, see e.g. BATT_MONITOR)

olliw42, 2019-02-12:
and to be consistent with COMPASS, BATT_MONITOR, BARO, GPS, ... one probably wants to rename RNGFND1_XXX to RNGFND_XXX

rmackay9, 2019-02-12:
I also noticed the issue with the first rangefinder being called RNGFND**1**_ instead of just RNGFND_

olliw42, 2019-02-12:
I probably should have added this
just delete the 1 in the strings here https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/RangeFinder.cpp#L49 and here https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/RangeFinder.cpp#L53
:)

rmackay9, 2019-02-12:
.. and there's probably a small change that needs to be made to the parameter conversion handling code as well..

olliw42, 2019-02-13:
to continue with improvement, note that these parameters

pin
ratiometric
powersave_range
stop_pin
settle_time_ms
scaling
offset
function

are used by ONLY AP_RangeFinder_analog.cpp and/or AP_RangeFinder_PX4_PWM.h. That is these should be moved to the backend, like it's done for the WASP rangefinder. Since these are 8 params, and there are 10 rangefinders, this would reduce the number of parameters MP and the user has to deal with by a staggering 80 !!! (which is probably nearly 10%). This would make it really much more practical to use 10 rangefinders.
IMHO.

(I've done that for my betacopter fork, and it really looks so much better, from a user perspective ;))

khancyr, 2019-02-13:
agree ! 80 less useless parameter is a big deal !

256shadesofgrey, 2019-02-13:
@olliw42 Copy pasta from my PR in regards to moving parameters to backend, since this was one of the first things that were considered:

The problem with adding parameters like `_ADDR` to the backend is that with the current implementation many backends require an instance of `AP_HAL::OwnPtr<AP_HAL::I2CDevice>` to be passed, which in turn requires an address which we don't have until we call `detect()`. To move parameters to the backends we would have to change the design of at least 5 already existing drivers, moving the `hal.i2c_mgr->get_device()` calls inside the drivers and making the `detect()` functions less "intelligent" (and at this point I'm starting to question why we need `detect()` at all).

So yeah, it would make it cleaner, but it's quite a bit of work for just aesthetics.

olliw42, 2019-02-13:
_ADDR was not in my list of params I suggested to move ;)

IamPete1, 2021-09-12:
closed by https://github.com/ArduPilot/ardupilot/pull/14560

closed 2021-09-12 by IamPete1