ArduPilot/ardupilot #10503: Reintroduce support for EagleTree Airspeed Sensor in a ArduPlane post PX4

JoshWelsh, 2019-02-12:
#### Issue details
When we left PX4 and moved on to something modern and efficient, support for the EagleTree airspeed sensor wasn’t brought over. Here is a white paper @magicrub found that should help: https://www.eagletreesystems.com/Manuals/microsensor-i2c.pdf

#### Version
ArduPlane master 

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Airplanes

#### Hardware type
Any

#### Logs
N/A

magicrub, 2019-02-12:
possible fix here: https://github.com/magicrub/ardupilot/tree/eagleTree_driver

rmackay9, 2019-02-12:
Tom on fire! :-)

magicrub, 2019-02-12:
The docs are absent about basic things like units of the data. I'm borrowing bits from the old PX4 driver so I hope it's right! I don't have hardware to test with

JoshWelsh, 2019-02-13:
Will try to get this bench tested in the next week. Hardware inbound. Thanks Tom!

Uravass, 2025-02-21:
Hi guys!  Does this work - I found an old unused eagle tree "airspeed expander with pitot" in my drawer and it would be nice to be able to use...