ArduPilot/ardupilot #10504: Copter: check motor thrust loss detector is identifying the problem motor correctly

rmackay9, 2019-02-12:
Below is a screen shot of a user log in which an external switch has been used to turn off motor 1 of Hexa-X vehicle.  The RCOU log makes it obvious that motor1 is the problem motor but the message indicates that the motor-thrust-loss-detectors thinks the problem motor is 4.  We should check whether this is a reporting error or if the detector has an issue causing it to misidentify the problem motor.

[image]

lthall, 2019-02-18:
https://github.com/lthall/Leonard_ardupilot/tree/20190218_MotorRedundancyFix

I believe this has happened because the aircraft is running with motor 4 high and when we detect thrust loss the filter has not let motor 1 climb above motor 4. By freezing it when thrust boost is enabled I believe we will fix this problem.

rmackay9, 2020-01-27:
I think we've resolved this issue in Copter-4.0.0 so closing

closed 2020-01-27 by rmackay9