ArduPilot/ardupilot #10510: Rover: "scan:695: scan past end of eeprom" displayed in SITL

rmackay9, 2019-02-13:
When rover is run is SITL the message, "scan:695: scan past end of eeprom" is displayed on the console.  I suspect this is being caused by some parameter conversion code that is using an incorrect parameter name.  I think it is innocuous but let's get rid of this message if possible.

[image]

khancyr, 2019-03-24:
I tried to solve this and the error append really on the start of the rover code, during rover object initialisation. Here is a gdb backtrace

````
Breakpoint 1, AP_Param::scan (target=0x7fffffffd6bc, pofs=0x7fffffffd6b6)
    at ../../libraries/AP_Param/AP_Param.cpp:695
695         Debug("scan past end of eeprom");
(gdb) backtrace
#0  AP_Param::scan (target=0x7fffffffd6bc, pofs=0x7fffffffd6b6)
    at ../../libraries/AP_Param/AP_Param.cpp:695
#1  0x00005555555d5041 in AP_Param::configured_in_storage (
    this=0x55555594b740 <rover+8384>)
    at ../../libraries/AP_Param/AP_Param.cpp:1217
#2  0x000055555557d618 in AP_Param::configured (
    this=0x55555594b740 <rover+8384>)
    at ../../libraries/AP_Param/AP_Param.h:427
#3  0x000055555557df30 in AP_ParamT<signed char, (ap_var_type)1>::set_default (
    this=0x55555594b740 <rover+8384>, v=@0x7fffffffd766: 0 '\000')
    at ../../libraries/AP_Param/AP_Param.h:676
#4  0x00005555555d86de in RangeFinder::RangeFinder (
    this=0x55555594b718 <rover+8344>, _serial_manager=..., 
    orientation_default=ROTATION_NONE)
    at ../../libraries/AP_RangeFinder/RangeFinder.cpp:157
#5  0x0000555555580077 in Rover::Rover (this=0x555555949680 <rover>)
    at ../../APMrover2/Rover.cpp:35
#6  0x000055555557602c in __static_initialization_and_destruction_0 (
    __initialize_p=1, __priority=65535) at ../../APMrover2/APMrover2.cpp:36
#7  0x000055555557619d in _GLOBAL__sub_I_hal ()
    at ../../APMrover2/APMrover2.cpp:329
#8  0x00005555556df8dd in __libc_csu_init ()
#9  0x00007ffff6ea7b28 in __libc_start_main (
---Type <return> to continue, or q <return> to quit---
    main=0x555555575642 <main(int, char* const*)>, argc=11, 
    argv=0x7fffffffd928, init=0x5555556df890 <__libc_csu_init>, 
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffd918)
    at ../csu/libc-start.c:266
#10 0x0000555555574a3a in _start ()

````
I don't understand why the parameter checking is failing on rover and not on other vehicle ...

WickedShell, 2019-04-05:
I'm pretty sure it's the selection of initialization order. The problem is the initializer for `_storage` in AP_Param has not been called yet, and Rangefinder is scanning for parameters in it's initializer.

PR #11016 (2019-04-05): Fix bad parameter reads on rover
This resolves #10510. What was happening is the rangefinder initializer was running before the AP_Param `_storage` initializer got to run. This lead to a `_storage.size()` of 0. The net effect is that we are unable to set the default orientation of the rangefinders. This is currently only happening on rover, but it depends entirely upon compiler decisions, and could easily happen on the other vehicles (or with the cross compiled output and we are just missing it).

The simple fix is to never try and manipulate a parameter during the constructor, which is what's [... cut]
closed 2019-04-05 by peterbarker