ArduPilot/ardupilot #10524: Copter: flowhold can lean past ANGLE_MAX parameter value

rmackay9, 2019-02-15:
This was [reported here](https://discuss.ardupilot.org/t/flowhold-allows-angles-larger-than-max-angle/38553).

it looks like there is a mistake on [this line](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_flowhold.cpp#L274).  The call to get_pilot_desired_lean_angles() is passing in the result of the "attitude_control-&gt;get_althold_lean_angle_max()" call twice instead of passing in aparm.angle_max.

peterbarker, 2020-06-18:
Looks like this went in when flowhold was first added.

Perhaps there was an intent - perhaps it was expected that the althold limit would always be less than the angle_max parameter?  Perhaps @tridge can remember?

abirdh, 2021-01-31:
Hello, I have this problem in copter 4.0.5.
Maybe someone know how can i fix this?
i'm experiencing this problem in flowhold mode (copter without gps). the copter have range finder.
the only problem is the lean angle that i can't configure

closed 2021-08-17 by rmackay9

Muhammetyilmaz7, 2024-08-13:
Hello, I am using dronekit python in quadcopter drone. This drone will perform a task autonomously. Let me briefly talk about the mission: The drone takes off and heads towards the specified location. After reaching a certain distance on the road, it must go to that location in a kamikaze dive manner. I am having trouble adjusting the pitch of the drone. How can I do this? How can the nose of the drone move to the desired position at a desired angle?

PR #18256 (2021-08-17): Copter: flowhold: use correct angle max
Fixes #10524

Flow hold was always using lean angle max, that is none configurable and can be very high if a copter has a low hover throttle.