ArduPilot/ardupilot #10541: Copter SITL build on OSX Mojave fails

bart, 2019-02-17:
### Bug report

**Issue details**

When running sim_vehicle.py command on OSX Mojave I'm getting an error.

**Version**
Copter-3.6.6

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

**Hardware type**
Pixhawk

**Logs**
```
...
[467/509] Compiling ArduCopter/landing_gear.cpp
Waf: Leaving directory `/Users/bart/Code/shiftAlpha/px4/ardupilot/build/sitl'
Build failed
 -> task in 'objs/AP_HAL_SITL' failed (exit status 1):
	{task 4423981088: cxx system.cpp -> system.cpp.0.o}
/usr/local/Cellar/gcc/8.2.0/bin/g++ -std=gnu++11 -fdata-sections -ffunction-sections -fno-exceptions -fsigned-char -Wall -Wextra -Wformat -Wshadow -Wpointer-arith -Wcast-align -Wundef -Wno-unused-parameter -Wno-missing-field-initializers -Wno-reorder -Wno-redundant-decls -Wno-unknown-pragmas -Werror=format-security -Werror=array-bounds -Werror=uninitialized -Werror=init-self -Werror=switch -Werror=sign-compare -Wfatal-errors -Wno-trigraphs -Werror=unused-but-set-variable -O3 -MMD -include ap_config.h -Ilibraries -Ilibraries/GCS_MAVLink -I. -I../../libraries -I../../libraries/AP_Common/missing -DSKETCHBOOK="/Users/bart/px4/ardupilot" -DCONFIG_HAL_BOARD=HAL_BOARD_SITL -DCONFIG_HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE ../../libraries/AP_HAL_SITL/system.cpp -c -olibraries/AP_HAL_SITL/system.cpp.0.o
SIM_VEHICLE: Build failed
SIM_VEHICLE: Killing tasks
```

`./waf configure` looks good:

```
./waf configure                                                                                                                           tags/ArduCopter-beta
Setting top to                           : /Users/bart/Code/shiftAlpha/px4/ardupilot
Setting out to                           : /Users/bart/Code/shiftAlpha/px4/ardupilot/build
Autoconfiguration                        : enabled
Setting board to                         : sitl
Checking for 'g++' (C++ compiler)        : /usr/local/Cellar/gcc/8.2.0/bin/g++
Checking for 'gcc' (C compiler)          : /usr/local/Cellar/gcc/8.2.0/bin/gcc
Checking for c flags '-MMD'              : yes
Checking for cxx flags '-MMD'            : yes
Checking for HAVE_CMATH_ISFINITE         : yes
Checking for HAVE_CMATH_ISINF            : yes
Checking for HAVE_CMATH_ISNAN            : yes
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes
Checking for header endian.h                   : not found
Checking for header byteswap.h                 : not found
Checking for program 'python'                  : /usr/local/opt/python@2/bin/python2.7
Checking for python version >= 2.7.0           : 2.7.15
Checking for program 'python'                  : /usr/local/opt/python@2/bin/python2.7
Checking for python version >= 2.7.0           : 2.7.15
Source is git repository                       : yes
Update submodules                              : yes
Checking for program 'git'                     : /usr/local/bin/git
Checking for program 'size'                    : /usr/bin/size
Benchmarks                                     : disabled
Unit tests                                     : enabled
Checking for program 'rsync'                   : /usr/bin/rsync
```

Do you have any idea what the reason could be? Thanks!

lvale, 2019-02-17:
Hi. Support questions are better handled at the forum.

Regarding your issue, check the correct GCC version for Ardupilot.

closed 2019-02-17 by lvale

bart, 2019-02-17:
All right, didn't know that. Thanks @lvale

bart, 2019-02-17:
Btw. it still doesn't work :(

lvale, 2019-02-19:
using these commands (also on Mojave 10.14.3)

```
 ./waf distclean
git submodule deinit -f . 
git clean -f -x -d 
git pull 
git submodule update --init --recursive 
./Tools/autotest/sim_vehicle.py -v ArduCopter -f heli -L TagusPark

```

[image]

ddomit, 2019-04-25:
> using these commands (also on Mojave 10.14.3)
> 
> ```
>  ./waf distclean
> git submodule deinit -f . 
> git clean -f -x -d 
> git pull 
> git submodule update --init --recursive 
> ./Tools/autotest/sim_vehicle.py -v ArduCopter -f heli -L TagusPark
> ```
> [image]

Where in the directory you have to be to be able to run this code?

peterbarker, 2019-04-25:
@ddomit the root directory of a git clone of ArduPilot.

ddomit, 2019-04-25:
THanks! I tried it but Im getting these errors:

[image]
[image]

peterbarker, 2019-04-25:
This PR fixes the build:  https://github.com/ArduPilot/ardupilot/pull/11094/commits

If you can't try that PR, please let me know and I'll start to push people on it :-)

aliabbasjaffri, 2019-04-29:
Running in the same issue here.
No idea why this was marked as closed!
[image]

srediske, 2020-04-15:
Ubuntu 18.04 with ArduCopter 4.0.3 (i guess, at least actual master).
I'm running everything in conda with python 2.7.15.

When running: `sim_vehicle.py -v ArduCopter -f gazebo-iris --map --console`
i get following error:

```
sim_vehicle.py -v ArduCopter -f gazebo-iris --map --console
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/sre/ZAL/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /home/sre/ZAL/ardupilot 
Setting out to                           : /home/sre/ZAL/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Using toolchain                          : native 
Checking for 'g++' (C++ compiler)        : /home/sre/anaconda3/envs/gym/bin/x86_64-conda_cos6-linux-gnu-c++ 
Checking for 'gcc' (C compiler)          : /home/sre/anaconda3/envs/gym/bin/x86_64-conda_cos6-linux-gnu-cc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
Checking for need to link with librt     : necessary 
Checking for library rt                  : yes 
Checking for feenableexcept              : yes 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for HAVE_MEMRCHR                      : yes 
Checking for program 'python'                  : /home/sre/anaconda3/envs/gym/bin/python 
Checking for python version >= 2.7.0           : 2.7.15 
Checking for program 'python'                  : /home/sre/anaconda3/envs/gym/bin/python 
Checking for python version >= 2.7.0           : 2.7.15 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /home/sre/anaconda3/envs/gym/bin/git 
Checking for program 'size'                    : /home/sre/anaconda3/envs/gym/bin/x86_64-conda_cos6-linux-gnu-size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : enabled 
Scripting runtime checks                       : enabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (2.424s)
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/sre/ZAL/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arducopter"
Waf: Entering directory `/home/sre/ZAL/ardupilot/build/sitl'
[457/682] Compiling libraries/AP_HAL_SITL/Storage.cpp
[459/682] Compiling libraries/AP_HAL_SITL/DSP.cpp
[468/682] Compiling libraries/AP_HAL_SITL/sitl_gps.cpp
[473/682] Compiling libraries/SITL/SIM_SerialRangeFinder.cpp
[474/682] Compiling libraries/SITL/SIM_Scrimmage.cpp
[475/682] Compiling libraries/SITL/SIM_Rover.cpp
../../libraries/AP_HAL_SITL/sitl_gps.cpp: In member function 'void HALSITL::SITL_State::_gps_nmea_printf(uint8_t, const char*, ...)':
../../libraries/AP_HAL_SITL/sitl_gps.cpp:653:6: error: '
   ' directive output may be truncated writing 2 bytes into a region of size between 1 and 3 [-Werror=format-truncation=]
 void SITL_State::_gps_nmea_printf(uint8_t instance, const char *fmt, ...)
      ^~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors

Waf: Leaving directory `/home/sre/ZAL/ardupilot/build/sitl'
Build failed
 -> task in 'objs/AP_HAL_SITL' failed (exit status 1): 
	{task 139998277248376: cxx sitl_gps.cpp -> sitl_gps.cpp.0.o}
[... 20 lines cut ...]
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for HAVE_MEMRCHR                      : yes 
Checking for program 'python'                  : /home/sre/anaconda3/envs/gym/bin/python 
Checking for python version >= 2.7.0           : 2.7.15 
Checking for program 'python'                  : /home/sre/anaconda3/envs/gym/bin/python 
Checking for python version >= 2.7.0           : 2.7.15 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /home/sre/anaconda3/envs/gym/bin/git 
Checking for program 'size'                    : /home/sre/anaconda3/envs/gym/bin/x86_64-conda_cos6-linux-gnu-size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : enabled 
Scripting runtime checks                       : enabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (2.384s)
```
Any ideas?

andyp1per, 2020-04-15:
You should use clang - works well for me

srediske, 2020-04-16:
You mean clang instead of gcc? I never switched that before and had never this issue. There must be an other problem or not?