ArduPilot/ardupilot #10543: Copter: Suddenly "Emergency Stop" when autotune and recover again.

gitgitgogogo, 2019-02-17:
### Bug report

**Issue details**
I set RC channel 11 for "Emergency Stop", and it worked well when I tested it on the ground.
But when the drone took off in Loiter mode and switch to autotune, everything was ok at the beginning, but after a while the drone suddenly fall down and touch the ground and then fly back to the sky again, so I made it land carefully and switch the RC channel 11 to HIGH to stop the motors.

I review the log and find there is a "Emergency Stop" in the sky and "Emergency Stop Cleared" after 0.1s, but I didn't switch RC channel 11 at that time, and the RCIN.C11 is normal in the log.

So what may cause the "Emergency Stop"?

**Version**
Arducopter v3.5.5

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
Pixhawk Cube
RC: Frsky Trains X9D+ (Firmware version: OpenTX v2.2.3)
RC MODULES: Frsky R9M (Firmware version: 190117)
Receiver: Frsky R9 (Firmware version:190201)

**Logs**
https://www.dropbox.com/s/rjy4aoqzglgqgu2/00000125.BIN?dl=0

[image]

IamPete1, 2019-02-17:
can you see if you can replicate the issue on the [current release](http://firmware.ardupilot.org/Copter/stable/CubeBlack/), 

Thanks

peterbarker, 2019-02-17:
On Sun, 17 Feb 2019, Peter Hall wrote:

> can you see if you can replicate the issue on the current release,

@gitgitgogogo there have been several fixes to RC-related issues - some 
exactly matching your symptoms - on the latest stable branch.

SergeyBokhantsev, 2019-02-18:
@peterbarker are you mean improvement for RC bounds checking for switches?

Whatever the root cause, log must show the reason of interlock. Now log shows only fact interlock was enabled but not the reason.
Indirectly I can assume there was a receiver glitch (because there is also CH4 spike at that moment). And for some reason the spike on CH11 was not logged (logging issue?) - is the log rate for RCIN is lower than real incoming data rate?

peterbarker, 2019-02-20:
On Mon, 18 Feb 2019, Sergey Bokhantsev wrote:

> (logging issue?) - is the log rate for RCIN is lower than real incoming data rate?

Yes.  50Hz vs 10Hz IIRC.

SergeyBokhantsev, 2019-02-20:
Then I believe we need a special event to be logged somewhere at https://github.com/ArduPilot/ardupilot/blob/61377040328ed2753193001a0df5441d3abb54d9/libraries/RC_Channel/RC_Channel.cpp#L488
Because RCIN log cannot reliable show the switch action by design.
The log event should contain channel and the switch position.

IamPete1, 2019-02-20:
That is what the "Emergency Stop" and "Emergency Stop Cleared" you saw in you log is.

peterbarker, 2019-02-20:
@SergeyBokhantsev I'm confused here.

Your initial report stated your were running old-stable - v3.5.5

I stated that there's been significant fixes in master to to with aux functions - in particular, adding the debouncing which is above the line you've referenced - in the master branch :-)

Curious: what happens when you simulate a radio-out even by turning the transmitter off?  Does it register a throttle failsafe, or otherwise get an RC failure?

SergeyBokhantsev, 2019-02-20:
@IamPete1 ok, I see set_motor_emergency_stop() is only called from init_aux() and do_aux() so we can say the reason was the RC channel switch.
It is just my thought this is not very good from the architectural standpoint because:
- set_motor_emergency_stop() is just a function that tomorrow could be called from another different places (scripts, new logic etc.) So the function does log the fact of the event, but not the event initiator source.
- it is possible that some aux functions have no EV logging. So they will not be logged.
So what I talking about is a generic logging for the aux_switch.

@peterbarker didn't know there is no debouncing in 3.5. I'm not a reporter of the issue, @gitgitgogogo to comment here

rmackay9, 2019-02-25:
I think we can close this now because auxiliary switch debouncing is addressed in Copter-3.6 (and later).

closed 2019-02-25 by rmackay9