ArduPilot/ardupilot #10545: Intergrated ROI camera to another mav

roque-canales, 2019-02-17:
Hello,

ROI is very interessant feature that automaticaly orient the camera through gimbal to GPS point.

There is possible to add AC3.6 UAV capability to ROI camera automaticaly to another mav connected trough mavlink on serial 2 port?

For example setting MNT_DEFLT_MODE to mavlink targeting and setting target MAV ID (parameter not available yet) of the mav that is the ROI ?

peterbarker, 2019-02-17:
On Sun, 17 Feb 2019, roque-canales wrote:

> There is possible to add AC3.6 UAV capability to ROI camera automaticaly to another mav connected trough mavlink on serial 2 port?

Not on 3.6, I'd guess.  Maybe for 3.7

> For example setting MNT_DEFLT_MODE to mavlink targeting and setting target MAV ID (parameter not available yet) of the mav that is the ROI ?

Well, we'd implement MAV_CMD_DO_SET_ROI_TARGET as a mavlink/mission 
message initially.

I have given this some thought but not acted on it.

roque-canales, 2019-02-25:
Hello Randy,

It will be a very cool feature, if it's possible to activate it without Following the target.

when you plan to release this feature?

rmackay9, 2019-02-26:
@roque-canales, no specific ideas on when we will get to implementing this feature I'm afraid.  I think it will happen eventually.. it's just hard to say when.

IamPete1, 2021-07-26:
Closed by, https://github.com/ArduPilot/ardupilot/pull/10830

closed 2021-07-26 by IamPete1