ArduPilot/ardupilot #10556: SBF driver unmounts disk rather than stops logging.

edwinhayes, 2019-02-18:
The Septentrio GPS driver controls logging by mounting and mounting the disk, rather than by turning the stream which is logging to disk on an off.  This might be convenient for users who want to remove the uSD after each flight, but breaks other systems, since the disk can also be accessed by other means (e.g. via ethernet).

A solution would be if the raw data logging parameter had an additional value, so there were values for 'log continuously', 'break up logs by stopping the stream' and 'break up logs by stopping the stream AND unmount the disk when it's not being used'.

jmachuca77, 2019-02-18:
@WickedShell

WickedShell, 2019-02-18:
This is by design. I actually originally checked with them if turning the stream off was sufficient to prevent disk corruption on loss of power, and the answer from their support was to unmount the disk rather then stop the stream. My *guess* for the reasoning is that:

1. The ArduPilot driver has no input on any of your other streams, which may also be logging to the disk, which may lead to ongoing disk activity still happening.
1. An unmounted disk should avoid any background updates/pushes to the file.
1. Mounting/unmounting the disk causes all streams that want to write to the disk to be smoothly reactivated.

It is an opt in feature. If you (or someone else) wants to implement and test the stream version I won't block it, but based on the official recommendation being to avoid the stream off approach and use the mounting commands instead, I can't say that it is something I will ever be investigating adding support for.

Final note: If you have M2 hardware the SD card can be remounted after unmounting, and you should be able to do that over your other interfaces as well. (The driver also supports that automatically when you try and arm again, but arming to write to an SD card is not a good/safe workflow for normal use of then fetching files)

edwinhayes, 2019-02-18:
Hmm, I guess it very much depends on what happens to the disk during a power cycle, assuming there aren't any streams writing to it.  If it's going to get wrecked anyway, then I guess unmounting it every time is the only safe choice.  I'll investigate this, and then if it seems like just stopping the streams would be safe, I'll put it on our list of dev tasks.

WickedShell, 2019-04-09:
Closing for now, as this was by design, and I haven't heard anything back. Please feel free to reopen/resume the discussion if you find out more.

closed 2019-04-09 by WickedShell