ArduPilot/ardupilot #10559: Hybrid RTL with Circle to ALT for QuadPlanes

ovolkmann, 2019-02-18:
Currently when invoking RTL on a quadplane with Q_RTL_MODE set to 1, "The initial altitude that will be aimed for in the fixed wing portion of the hybrid RTL is the same as for a fixed wing RTL. You should set your rally point altitude and ALT_HOLD_RTL options appropriately to ensure that the aircraft arrives at a reasonable altitude for a vertical landing. A landing approach altitude of about 15 meters is good for many QuadPlanes. This should be greater than or equal to the Q_RTL_ALT values."

The problem is that if the plane is too close that the pitch angle does not allow the plane to reach the target altitude of "ALT_HOLD_RTL" or the Rally Point Altitude, it will transition as soon as it is in WP_RADIUS. This results in the plane transitioning to copter mode and having a potentially long way to descend before landing which can be a safety risk especially on larger platforms that are underpowered on the VTOL side. 

I'd like to propose a version of Hybrid RTL that functions just as it does now, but takes altitude into account before transition by means of circling until a specific RTL Transition Altitude is reached. For example, RTL invoked, plane flies towards rally point or ALT_HOLD_RTL target altitude and position, if target altitude is not reached, circle/loiter in plane mode down to target altitude, invoke transition, QRTL -> LAND.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ Y ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

pompecukor, 2019-02-18:
Not exactly what you described, but solves your request too. So I think this can be closed:

https://github.com/ArduPilot/ardupilot/pull/9707

ovolkmann, 2019-02-24:
Looks like it's already under development in other requests. Closing as is redundant. Thanks!

CallanDaniel, 2021-04-22:
@ovolkmann i see this was closed, but this funcionality still is'nt availble when returning to land, only in AUTO with a mission planned. What if there is no mission planned and you invoke RTL from a high altitude? I'm just interested to know how you are handling this with current code?

tridge, 2021-04-22:
re-opened for more discussion

CallanDaniel, 2021-04-22:
Great, thanks @tridge

pompecukor, 2021-04-22:
@CallanDaniel @tridge it is solved in another PR that we had Peter do for us. Let me link it. It works very well to :)

pompecukor, 2021-04-22:
https://github.com/ArduPilot/ardupilot/pull/16977

IamPete1, 2021-07-26:
VTOL approach from RTL now possible, https://github.com/ArduPilot/ardupilot/pull/16977 

I think that is more or less exactly what was wanted, so I'm going to close. 

Feel free to re-open if I's wrong.

closed 2021-07-26 by IamPete1