ArduPilot/ardupilot #10563: Invalid INS_GYRO_FILTER parameter range

p2041, 2019-02-19:
### Bug report

**Issue details**
Hello.

I found that, if INS_GYRO_FILTER is set as 1, a drone in SITL shows very unstable stability. 
I also found that 1 is a valid according to the document. http://ardupilot.org/copter/docs/parameters.html#ins-gyro-filter-gyro-filter-cutoff-frequency

Please note, unlike this issue (#10435), it does not cause division-by-zero --- just unstable control.

**Version**
ArduCopter 3.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[O ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
Probably, other platforms may show the same problems...?

**Airframe type**
Quadcopter

**Hardware type**
SITL

rmackay9, 2019-02-19:
There are a lot of parameters that will cause the SITL vehicle to become unstable because of the virtual vehicle's physics.  These values might be valid on other vehicles though.  At some point, reducing the acceptable range will reduce the physical size (either small or large) of vehicles we can support.

Still, perhaps the lower range should be increased.

IamPete1, 2021-08-29:
The trouble is we can't set two ranges, lots of these params include 0 as a disable. I guess we could have a range and a values. If we did want to do something like that we should have a issue for all params not a issue per param and probbibly would need a GCS change to not warn if the value is ouside the range if it is in the @ values.

In anycase I think I'm going to close this, there are some vehicles where a low value might be valid, the param is marked for advanced users.

closed 2021-08-29 by IamPete1