ArduPilot/ardupilot #10564: Invalid INS_POSX_Z parameter

p2041, 2019-02-19:
### Bug report

**Issue details**
Hello,
I found that INS_POS1_X, INS_POS2_X, INS_POS3_X, INS_POS1_Y, INS_POS2_Y, INS_POS3_Y are safe even if they are changed into a very strange value. (Due to the EKF? maybe).

However, I found that INS_POS1_Z, INS_POS2_Z, INS_POS3_Z still show very bad performance with strange values.
In addition, if they are set as ranged values in the document (-10 ~ 10), a drone shows the same problem. 
It works approximately with -1.5 ~ 2.5. 
If the other parameter (e.g.,  both INS_POS1_Z and INS_POS2_Z), it works with a smaller range (about -0.1 ~ 2.5).
Because it works with a very small value and the size of drones may not be 10 meters, I think their ranges may be smaller than the current ranges. 

**Version**
ArduCopter  3.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[O] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
SITL

p2041, 2019-02-19:
@rmackay9 Please note the following things.
For XY, I found that a SITL drone can recover its state using EKF after some times (e.g., 10 sec?).
I think EKF can ignore or handle X, Y axes problems.
However, for Z-axis, a SITL drone cannot recover it. That's why I wrote only "Z" axis.

rmackay9, 2019-02-20:
@p2041, I wonder if the problem you found is because the SITL copter's IMU is in the middle or is it because of a general limitation of the EKF.  I suspect it is the former.. so the user must be careful to set the Z-axis correctly but that is not really a problem.

IamPete1, 2021-08-08:
I think this is expected, if you set the IMU offsets so they don't match the physical vehicle the EKF will have a hard time.

closed 2021-08-08 by IamPete1